pybamm-team/PyBaMM #1264: Use of WLTC drive cycle

maurosgroi, 2020-12-01:
Dear developers,
I'm trying to use a WLTC cycle for heavy duty vehicles as a test cycle. The cycle lasts more than 2800 s and is sampled every 100 ms.
Essentially I'm substuting the original csv file in tutorial 4 with mine but the solution is cut at about 770 s.
Is there any limitation in the number of point that can be imported?
My csv file for reference is uploaded here: https://drive.google.com/file/d/1MJBdq-g5RjednvSJ8e-LMkQnnxe1OWvj/view?usp=sharing 

I'm reporting my script below:

import pybamm
import pandas as pd # needed to read the csv data file
chemistry = pybamm.parameter_sets.Chen2020
parameter_values = pybamm.ParameterValues(chemistry=chemistry)
model = pybamm.lithium_ion.SPM()
drive_cycle = pd.read_csv("D:/WLTC.csv", comment="#", header=None).to_numpy()
timescale = parameter_values.evaluate(model.timescale)
current_interpolant = pybamm.Interpolant(drive_cycle, timescale * pybamm.t)

parameter_values["Current function [A]"] = current_interpolant
sim = pybamm.Simulation(model, parameter_values=parameter_values)
sim.solve(t_eval=None)
sim.plot(["Current [A]", "Terminal voltage [V]"], time_unit='seconds', variable_limits='tight')

Thanks a lot in advance and best regards,
Mauro Sgroi.

tinosulzer, 2020-12-01:
Is this WLTC (as in the filename) or WLTP (as in the comment in the file)?

tinosulzer, 2020-12-01:
The simulation is stopping because the voltage is hitting its maximum cut-off point. You can see this by setting
```python3
pybamm.set_logging_level("INFO")
```
at the start of your script.
Maybe you should be discharging with WLTC instead of charging, you can do this as follows to see that this works:
```python3
drive_cycle = pd.read_csv(
    "D:/WLTC.csv", comment="#", header=None
).to_numpy()
drive_cycle[:, 1] = -drive_cycle[:, 1]
```
If you do mean to charge instead of discharge, you can change the initial conditions to do this.
Also be sure to check that the currents in your drive cycle are appropriate for the capacity of the battery your are simulating (in your parameter set)

maurosgroi, 2020-12-01:
Thanks a lot for the help!

The cycle is a WLTC for heavy duty derived by the absorption profile of a real vehicle. Sorry sometimes I uses P (=procedure) and C (=cycle) as synonyms.

The profile was scaled down at cell level and the package is composed by 18650 cells. So I think that the cycle should be not too demanding for the Chen2020  dataset. My intention is to prepare my own dataset for the cells under study.

What about the sign convention for the current in PyBamm? 
On my the cell cycling facilities the current is considered  positive if supplied to the cell. 

Best regards,
Mauro.

tinosulzer, 2020-12-01:
In PyBaMM the convention is the opposite. A positive current means discharge (current taken from the cell), and a negative current means charge (current supplied to the cell).

maurosgroi, 2020-12-01:
I've tried to change the sign of the current of the cycle and it work properly.
Thanks a lot!
Best regards,
Mauro.

tinosulzer, 2020-12-01:
Can we add WLTC to the standard drive cycles in pybamm?

maurosgroi, 2020-12-01:
Yes for sure. 
But for each application the current profile on the will depend on the power absorption  of the electric engine of the vehicle and on the configuration of the battery package. 

Maybe I can provide something that is representative of a real vehicle with a certain battery package configuration as an example.

tinosulzer, 2020-12-01:
Yes, that would be great. It can just be a representative example, with some information (e.g. in a separate markdown file - we could start a README.md for the drive cycles folder) about how it was obtained, and which chemistry / parameter set it is valid for

maurosgroi, 2020-12-01:
Yes, ok. I will work on it.
Best regards,
Mauro.

tinosulzer, 2020-12-02:
Yes. This is how you would do a 4W constant-power discharge:
```python3
import pybamm
import numpy as np

pybamm.set_logging_level("INFO")

# load models
model = pybamm.lithium_ion.SPM({"operating mode": "power"})

# load parameter values and process models and geometry
param = model.default_parameter_values

# 4W discharge
param.update({"Power function [W]": 4}, check_already_exists=False)

sim = pybamm.Simulation(model, parameter_values=param)
t_eval = np.linspace(0, 3600, 100)
sim.solve(t_eval)

sim.plot()
```
so just replace the 4 in `"Power function [W]": 4` with your power interpolant and it should work. It might a bit slower as it will be a DAE model, instead of an ODE model, and DAE models struggle a bit with drive cycles

maurosgroi, 2020-12-02:
It works.
But now the cell is too small for the cycle... Since the cycle that I'd like to import was done on a 55Ah NMC/C cell.
I'm working on it.
Otherwise I can try to prepare a datafile fit for a 18650 cell..

Best regards,
Mauro.

maurosgroi, 2020-12-03:
Dear all',
Im posting here the code to import the WLTC cycle.
The file contains the data of power absorption in kW obtained on the cycle using a vehicle with a mass of 1500 kg and an electric powertrain with a maximum power of 85 kW. The sampling is every 100 ms.

The cycle can be downloaded here: https://drive.google.com/file/d/1MJBdq-g5RjednvSJ8e-LMkQnnxe1OWvj/view?usp=sharing

And the code to import here is below. It is necessary to make an hypotesis on the battery package. In the example I'm using a package made of 2000 18650 cells with a capacity of 3Ah connected in series and parallel (100sx20p).

import pybamm
import numpy as np
import pandas as pd

pybamm.set_logging_level("INFO")

# load models
model = pybamm.lithium_ion.SPM({"operating mode": "power"})

# load parameter values and process models and geometry
param = model.default_parameter_values

drive_cycle = pd.read_csv("D:/WLTC.csv", comment="#", header=None).to_numpy()

""" Scale down down the power to cell level
Note that the supplied profile represents the power absorption 
of a vehicle with a power of 85 kW and mass 1500 kg on the WLTC cycle.
,
1) Convert kW to W (multiply by 1000)
2) Hypotesis on the battery package: 
18650 cell with 3 Ah capacity
100 cells in series x 20 parallels: 2000 cells (divide by 2000)
"""
conv=1000 # kW to W
series=100
parallel=20

drive_cycle[:, 1] = conv*drive_cycle[:, 1]/(series*parallel) # Power on the cell

# Create interpolant
timescale = param.evaluate(model.timescale)
power_interpolant = pybamm.Interpolant(drive_cycle, timescale * pybamm.t)

# discharge
param.update({"Power function [W]": power_interpolant}, check_already_exists=False)

sim = pybamm.Simulation(model, parameter_values=param)
t_eval = np.linspace(0, 1800, 18001)
sim.solve(t_eval)

sim.plot(["Current [A]", "Terminal voltage [V]"], time_unit='seconds', variable_limits='tight')

tinosulzer, 2020-12-03:
Thanks @maurosgroi , could you fork the repository and make a pull request? 
- the drive cycle(s) should be added to `pybamm/input/drive_cycles`
- the script should be added to `examples/scripts/`

More detailed instructions for contributing are here: https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#workflow

maurosgroi, 2020-12-03:
Dear @tinosulzer,
I'm not confident wtih github.
So how should I proceed to fork the repository?
Best regards,
Mauro.

tinosulzer, 2020-12-03:
No worries. In the contributing file, there are links with detailed instructions on how to do each step. For example, forking is explained [here](https://docs.github.com/en/free-pro-team@latest/github/getting-started-with-github/fork-a-repo#fork-an-example-repository)

It might be a little bit intimidating at first, but you should get the hang of it pretty quickly 🙂

Saransh-cpp, 2021-02-03:
Hi there, can I work on this issue if no one is working on it?

maurosgroi, 2021-02-03:
@Saransh-cpp,
do you intend to open the fork?
It would be very useful.
Best regards,
Mauro.

Saransh-cpp, 2021-02-03:
@maurosgroi I can quickly guide you with the basics if you want to add your code in the repository. You must have git installed on your machine.

1. Fork
[image]

The right most option which says 'fork' after opening the pybamm repository is the one for forking. Clicking it will create a fork (or a duplicate repository) in your profile.

2. Clone the forked repository
[image]

You will get this clone button in your forked repository which you can access through your GitHub profile, which you can clone on your machine using
```git clone <the link which appears in the clone menu of your forked repository>```

3. After this you can find the detailed explaination of how to install and how to contribute [here](https://pybamm.readthedocs.io/en/latest/?badge=latest) which is the website for docs of pybamm.

Saransh-cpp, 2021-02-05:
@maurosgroi any updates? 

Also I had some questions with the code
```
drive_cycle[:, 1] = convdrive_cycle[:, 1]/(seriesparallel) # Power on the cell
```
In this line, is it supposed to be ```series*parallel``` and it gives me an error on ```convdrive_cycle```, is it supposed be initialised first?

tinosulzer, 2021-02-05:
@all-contributors add @maurosgroi for ideas

maurosgroi, 2021-02-08:
@Saransh-cpp,
I confirm that there is an error.
The correct line is:

drive_cycle[:, 1] = conv*drive_cycle[:, 1]/(series*parallel) # Power on the cell

In my fork it is already uploaded correctly.

Best regards,
Mauro.

Saransh-cpp, 2021-02-08:
@maurosgroi oh great! You will get used to Git and GitHub pretty quick. Good luck!:) and don't forget to create a pull request once you are done coding in your fork:)

ssingh-ipa, 2021-05-10:
Hello,

Is there any update on the code to import the WLTC cycle? I am getting the following error:
1.     power_interpolant = pybamm.Interpolant(drive_cycle, timescale * pybamm.t)
TypeError: __init__() missing 1 required positional argument: 'children'
2.     drive_cycle[:, 1] = convdrive_cycle[:, 1]/(seriesparallel) # Power on the cell
NameError: name 'convdrive_cycle' is not defined

I am trying to use pybamm for the first time in order to execute lithium battery models with real world driving cycle information.

Saransh-cpp, 2021-05-10:
Hi @ssingh-ipa, it should be `conv*drive_cycle[:, 1]/ series*parallel`, and I think for the `power_interpolant` it is supposed to be -
```py
power_interpolant = pybamm.Interpolant(drive_cycle[:, 0], drive_cycle[:, 1], timescale * pybamm.t)
```
You can checkout the [change-input-current](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/change-input-current.ipynb) example notebook.

ocsafeer, 2022-11-23:
Hi What is the least possible time step in the drive cycle

cyasing, 2023-01-20:
> The file contains the data of power absorption in kW obtained on the cycle using a vehicle with a mass of 1500 kg and an electric powertrain with a maximum power of 85 kW. The sampling is every 100 ms.

@maurosgroi Could you explain how you converted the original WLTC data from speed vs time to power vs time? I want to simulate my own configuration of vehicle.

maurosgroi, 2023-01-22:
Dear Cyasing,
the power was measured on a real segment A vehicle using a chassis
dynamometer.
Maybe you can find in the literature similar power profiles of your
interest.
Best regards,
Mauro.

cyasing, 2023-01-23:
> 

I see. Thanks!

cyasing, 2023-01-23:
> Dear Cyasing, the power was measured on a real segment A vehicle using a chassis dynamometer. Maybe you can find in the literature similar power profiles of your interest. Best regards, Mauro.

I see, thanks!