pybamm-team/PyBaMM #1265: Unable to Update Initial Conditions for Simulation

alibh95, 2020-12-02:
I am trying to simulate a driving cycle followed by CCCV charging.  Driving cycles are simulated by creating a current function parameter. And CCCV charging is simulated using Experiment class. To do so, I am solving the simulation with current function and trying to concatenate initial conditions to the model  and again starting the simulation with the Experiment for CCCV Charging. 

**Code**
`simulation1 = pybamm.Simulation(model, parameter_values=param, solver=solver)`
`simulation1.solve() #Driving Cycle`
`model.concatenated_initial_conditions = pybamm.Vector(solution1.y[:, -1][:, np.newaxis])`
`simulation2= pybamm.Simulation(model, experiment=experiment, solver=pybamm.CasadiSolver())`
`simulation2.solve() #CCCV Charging`

**Issue**
Unable to Update Initial Conditions for Simulation using 
`model.concatenated_initial_conditions = pybamm.Vector(solution1.y[:, -1][:, np.newaxis])`
Both Simulations starts from default initial conditions and at time t=0.

**Expected behaviour**
simulation2 must start with the initial conditions and time at the end of the simulation1

tinosulzer, 2020-12-02:
Yeah, this should work but unfortunately doesn't right now because of the way `model.new_copy` is currently set up.
Keep an eye on:
#1205 for a fix to new_copy
#1194 for a workaround on updating initial conditions (this is probably the easiest fix)
#1193 to allow drive cycles directly in experiments so you don't need to re-initialize

Any one of these would fix your problem or provide a workaround, so I'm going to close this issue as duplicate. If you are willing to help that would be great, but unfortunately these aren't the most beginner-friendly issues.

closed 2020-12-02 by tinosulzer

ssingh-ipa, 2021-06-27:
Hello,

I cannot follow if the issue has been resolved or now. 
As mentioned above, in order to do the following:
"simulation2 must start with the initial conditions and time at the end of the simulation1", 
what is the method of doing this for now? Could you please give an example showing how to implement #1193, if that's the way for going forward with this? 

Thank you.

tinosulzer, 2021-06-28:
`simulation2.solve(starting_solution=simulation1.solution)`

ssingh-ipa, 2021-07-02:
"ValueError: starting_solution can only be provided if simulating an Experiment."

I am solving almost the same problem as mentioned in the example above. First simulation is cccv_charging and the second simulation is a driving cycle. starting_solution does not work in this case. Please let me know if the following is technically doing the same, i.e. updating and executing the simulation. Is there a better solution? 

model = model.set_initial_conditions_from(sol_cccv, inplace=True)
solution = sim_cccv.solve() + sim.solve() 
#sim is the driving cycle here. I had to manually set the starting time for the sim driving cycle to 3601.