pybamm-team/PyBaMM #1275: Adding new model

Malipour14, 2020-12-04:
**Summary**
 Adding a new model to PyBaMM. 

 **Motivation**
Hello, I want to develope a new lumped-electrochemical model (an electrochemical model with lumped parameters) reusing the pybamm. Before starting, I wanted to now if it is doable. Thanks a lot.

tinosulzer, 2020-12-05:
Hopefully. What is the model? Do you have a reference?

Malipour14, 2020-12-05:
It is a kind of 0-order model that variables are only time-dependent which is so called "Lumped-electrochemical model"

https://doi.org/10.3182/20100712-3-DE-2013.00010

tinosulzer, 2020-12-05:
That looks interesting, thanks. We already have those particle, potential, and temperature submodels implemented. It would be great if you could add the electrolyte model in Section 3.2 as a new submodel, then you can reuse the other parts of the code. Let me know if you are interested in doing that and I can help you get started

Malipour14, 2020-12-05:
Thank you for your response. Yes, I'm totally interested to do it.
I'll be grateful if you help me with the procedure.
How the sub-models are coupled then?

tinosulzer, 2020-12-05:
Unfortunately we don't yet have good documentation explaining how the submodels are coupled. Basically, you need to add a new file to [this folder](https://github.com/pybamm-team/PyBaMM/tree/develop/pybamm/models/submodels/electrolyte_diffusion), using for example the "full diffusion" file to get you started.
Once you set it all up properly, you can add an option to call your new submodel from one of the model classes, e.g. [SPMe](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/lithium_ion/spme.py). Then, all of the other submodels will use the electrolyte concentration that is defined by your new submodel, and you don't need to change anything else.
If you're not familiar with defining equations in the pybamm format, I recommend [these tutorials](https://github.com/pybamm-team/PyBaMM/tree/develop/examples/notebooks/Creating%20Models)

Malipour14, 2020-12-06:
Thanks for the information. 
Is it possible to collect all the supermodels in one full model and use it as a new model?

Malipour14, 2020-12-06:
The reason why I want to develop the full model form the beginning is that the existing sub-models for SPMe or SPM models are not developed based on the lumped-parameters. For example, in the lumped-model, Fick's law for diffusion is not  valid since the concentration in the solid phase is a function of time only and not location (x or r)! therefore, the whole model should be written from the beginning.

tinosulzer, 2020-12-06:
In the case of the particles you can use the various polynomial approximation models from Subramaniam 2005 that we have already implemented, for example [here](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/compare-particle-diffusion-models.ipynb).
If you really want to make a full model yourself, check out the [BasicSPM](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/lithium_ion/basic_spm.py) for a template, but we don't want to add too many of these as there are exponentially many different combinations of submodels possible, and it becomes hard to maintain.

Malipour14, 2020-12-15:
Hi,

Could you please look at the following code. I do not know what is the
problem.
I got this error: model is underdetermined (too many variables). Thank
you. Regards.

Mohammad Alipour, Ph.D.
Engineering College
Koç University

Office:ENG106

Rumelifeneri Yolu, 34450 Sarıyer İSTANBUL

#Lumped_Electrochemical_Model

import pybamm
import pandas as pd
import numpy as np
import matplotlib.pyplot as plt
import sympy as syp

model=pybamm.BaseModel()

###########
#Parameters
a_s_p= pybamm.Parameter("Positive Electrode Specific Surface Area [m−1]")
a_s_n= pybamm.Parameter("Negative Electrode Specific Surface Area [m−1]")
A_p  = pybamm.Parameter("Positive Electrode Area [m2]")
A_n  = pybamm.Parameter("Negative Electrode Area [m2]")
alpha_p = pybamm.Parameter("Cathodic Transfer Coefficient")
alpha_n = pybamm.Parameter("Anodic Transfer Coefficient")
c_s_p_max = pybamm.Parameter("Maximum Concentration in Positive Electrode [mol.m-3]")
c_s_n_max = pybamm.Parameter("Maximum Concentration in Negative Electrode [mol.m-3]")
c_e_0 = pybamm.Parameter("Initial Electrolyte Concentration [mol.m-3]")
Ds_p = pybamm.Parameter("Diffusion Constant in Positive Particle [m2.s-1]")
Ds_n = pybamm.Parameter("Diffusion Constant in Negative Particle [m2.s-1]")
D_e  = pybamm.Parameter("Diffusion Constant in Electrolyte [m2.s-1]")
eps_e_p = pybamm.Parameter("Electrolyte Volume Fraction in Positive Electrode")
eps_e_n = pybamm.Parameter("Electrolyte Volume Fraction in Negative Electrode")
F    = pybamm.Parameter("Faraday Constant [C.mol-1]")
k_eff_p = pybamm.Parameter("Effective Diffusive Conductivity in Positive Electrode [S.m-1]")
k_eff_n = pybamm.Parameter("Effective Diffusive Conductivity in Negative Electrode [S.m-1]")
k_eff_s = pybamm.Parameter("Effective Diffusive Conductivity in Seperator [S.m-1]")
k_p   = pybamm.Parameter("Reaction Rate at Positive Electrode [mol-0.5.m2.5.s-1]")
k_n   = pybamm.Parameter("Reaction Rate at Negative Electrode [mol-0.5.m2.5.s-1]")
L_p  = pybamm.Parameter("Positive Electrode Thickness [m]")
L_n  = pybamm.Parameter("Negative Electrode Thickness [m]")
L_s  = pybamm.Parameter("Seperator Thickness [m]")
Rs_p = pybamm.Parameter("Positive Particle Radius [m]")
Rs_n = pybamm.Parameter("Negative Particle Radius [m]")
R    = pybamm.Parameter("Universal Gas Constant [J.mol-1.K-1]")
t_0  = pybamm.Parameter("Transfer Number of Li+")
T    = pybamm.Parameter("Temperature [K]")

##########
#Variables
Q     = pybamm.Variable("Discharge capacity [A.h]")
c_s_p = pybamm.Variable("Volume Avereged Solid Phase Concentration [mol.m-3]")
c_s_n = pybamm.Variable("Volume Avereged Solid Phase Concentration [mol.m-3]")
c_e_p = pybamm.Variable("Volume Avereged Electrolyte Phase Concentration [mol.m-3]")
c_e_n = pybamm.Variable("Volume Avereged Electrolyte Phase Concentration [mol.m-3]")
i_0_p = pybamm.Variable("Exchange Current Density in Positive Electrode [A.m-2]")
i_0_n = pybamm.Variable("Exchange Current Density in Negative Electrode [A.m-2]")
j_p   = pybamm.Variable("Intercalation Current Density in Positive Particle[A.m-3]") 
j_n   = pybamm.Variable("Intercalation Current Density in Negative Particle [A.m-3]")
eta_p = pybamm.Variable("Overpotential in Positive Electrode [V]")
eta_n = pybamm.Variable("Overpotential in Negative Electrode [V]")
theta_p = pybamm.Variable("O")
[... 100 lines cut ...]
        "Effective Diffusive Conductivity in Positive Electrode [S.m-1]":100,
	"Effective Diffusive Conductivity in Negative Electrode [S.m-1]":11,
	"Effective Diffusive Conductivity in Seperator [S.m-1]":100,
        "Reaction Rate at Positive Electrode [mol-0.5.m2.5.s-1]":1e-11,
	"Reaction Rate at Negative Electrode [mol-0.5.m2.5.s-1]":1e-11,
        "Positive Electrode Thickness [m]":70e-6,
	"Negative Electrode Thickness [m]":70e-6,
	"Seperator Thickness [m]":25e-6,
	"Positive Particle Radius [m]": 10e-6,
	"Negative Particle Radius [m]": 10e-6,
        "Universal Gas Constant [J.mol-1.K-1]":8.314,
        "Transfer Number of Li+":0.23,
        "Temperature [K]":303
        }
    )

#disc = pybamm.Discretisation()
#disc.process_model(model);

param.process_model(model)

# solve
dae_solver = pybamm.CasadiSolver(mode="safe") # use safe mode so that solver stops at events
t = np.linspace(0, 2, 10)
solution = dae_solver.solve(model, t)

tinosulzer, 2021-09-09:
@Malipour14 any update on this?

closed 2021-09-14 by tinosulzer