pybamm-team/PyBaMM #1299: Calculation Incremental capacity

maurosgroi, 2020-12-14:
Dear developers,
I'd like to calculate with Pybamm the incremental capacities curves dQ/dV versus V.

This would help to interpret and analyse experimental data related to the ageing of the Li-ion cells.

I'm trying to calculate the charge and use numpy to calculate dQ/dV but I cannot properly convert the pybamm.solvers.processed_variable.ProcessedVariable (for example, the cell potential, the time and the current) to numpy arrays.

How should I proceed?

Or this feature is already implemented?

Thanks a lot and best regards,
Mauro Sgroi.

tinosulzer, 2020-12-14:
Hi Mauro,
To access the numpy arrays, you can do e.g. `solution["Time [h]"].data`. Is this what you are looking for?
Are you trying to do dQ/dV on the output of PyBaMM simulations, or on experimental data?

maurosgroi, 2020-12-14:
Dear Valentin,
Thanks a lot.
It seems to be working. 
I'm trying to calculate the ICA of the output of PyBaMM simulations. 
To compare it with the experimental one.
Best regards,
Mauro.

closed 2020-12-15 by tinosulzer