pybamm-team/PyBaMM #1308: Error while simulating a charging protocol

optibikash, 2020-12-26:
**Describe the bug**
The following error comes when I try to implement the pulse charging with discharge. 
2020-12-26 09:57:56,970 - [INFO] casadi_solver._integrate(145): Start solving Doyle-Fuller-Newman model with CasADi solver with 'safe without grid' mode
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
Traceback (most recent call last):
  File "/home/apps/iitg/ANACONDA3/2020/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/home/apps/iitg/ANACONDA3/2020/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/scratch/binita.nanda/Bikash_Python_works/With_discharge/Set1/Puse_Charging_with_discharge.py", line 75, in <module>
    sim.solve(solver=pb.CasadiSolver(mode="safe without grid", dt_max=120))
  File "/home/binita.nanda/.local/lib/python3.8/site-packages/pybamm/simulation.py", line 426, in solve
    self.step(
  File "/home/binita.nanda/.local/lib/python3.8/site-packages/pybamm/simulation.py", line 488, in step
    self._solution = solver.step(
  File "/home/binita.nanda/.local/lib/python3.8/site-packages/pybamm/solvers/base_solver.py", line 802, in step
    solution = self._integrate(model, t_eval, ext_and_inputs)
  File "/home/binita.nanda/.local/lib/python3.8/site-packages/pybamm/solvers/casadi_solver.py", line 201, in _integrate
    raise pybamm.SolverError(
pybamm.expression_tree.exceptions.SolverError: Maximum number of decreased steps occurred at t=3648000.3156365054. Try solving the model up to this time only or reducing dt_max (currently, dt_max=7.5).
[binita.nanda@gpu-l

**To Reproduce**
Write code for pulse charging with discharge and run the code with the following options:
options = {"sei": "ec reaction limited", "sei porosity change": True, "thermal": "x-full"}
param = pb.ParameterValues(chemistry=pb.parameter_sets.Ramadass2004)
*Experiment is a list described in pulse_w_d* 
experiment = pb.Experiment(pulse_w_d*300)
sim = pb.Simulation(model, experiment=experiment, parameter_values=param)
sim.solve(solver=pb.CasadiSolver(mode="safe without grid", dt_max=120))
sol = sim.solution

**Expected behaviour**
The code runs for around 150 cycles and then gives the error described above.

**Screenshots**
Error is the same as copied above. 

**Additional context**
Add any other context about the problem here.

rtimms, 2020-12-31:
Could you try solving for a shorter time (up to the time given in the error message at which the maximum number of decreased steps occurred) and plotting the solution? My guess is that there is something in the model making it hard to solve here (e.g. getting close to zero concentration). The solver is having to work too hard and gives up. There might be some solver options you can play with to help, but it would be good to take a look at the solution first.

tinosulzer, 2021-03-29:
Can you try with `CasadiSolver("fast with events")`?

Also, to debug this faster, you should run up to the save solution that works, then save the last cycle (`solution.cycles[-1].save(filename)`), then load that cycle (`pybamm.load(filename)`) and set it as the `starting_solution` when you call `sim.solve`

optibikash, 2021-04-01:
Hi @tinosulzer . Thanks for remembering the problem and suggesting the solution. I used CasadiSolver("fast with events") to simulate the model. The simulation ran with no errors.

closed 2021-04-01 by tinosulzer

gupspriya, 2025-01-22:
Hi, I am using PyBaMM for LLI loss with respect to cycles. I am running for 1000 cycles but after around 600 cycle the model is giving error mentioned below:
At t = 100.79 and h = 2.04493e-10, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 243.462 and h = 2.36791e-10, the corrector convergence failed repeatedly or with |h| = hmin.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2025-01-22 16:00:31.868 - [ERROR] callbacks.on_experiment_error(233): Simulation error: Maximum number of decreased steps occurred at t=8910230.814540884 (final SolverError: 'Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1432:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:361:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.'). For a full solution try reducing dt_max (currently, dt_max=300) and/or reducing the size of the time steps or period of the experiment. Set `return_solution_if_failed_early=True` to return the solution object up to the point where failure occured.

I tried some solutions as suggested by contributors in this issue as well as in other similar issues, for example,
1. I have tried by reducing dt_max.
2. I have tried by changing the period of experiment.
3. I have tried all modes of Casadisolver.
4. I have tried IDAKUSolver.

I am getting the same error all the time. It will be a great help if you can suggest some other possible solutions to resolve this issue