pybamm-team/PyBaMM #1311: Homogenised jelly roll models

rtimms, 2020-12-29:
Add some of the models from the paper [Homogenisation of spirally-wound high-contrast layered materials](https://arxiv.org/abs/2011.04187), S. Psaltis, R. Timms, C.P. Please, S.J. Chapman. 

- [ ] Add the “reasonably/very conductive” model for the electrical problem. This is the limit you are normally in since the electrical conductivity of the current collectors is so large compared to that of the other components. This model is similar to the “1+1D” pouch cell model, but here you solve a 1D problem in `r` for the current collector potentials (note the operator is not the usual laplacian since the geometry is important here - see paper for details). In this limit the potentials do not depend on `theta`. 
- [ ] Add the “poorly/reasonably conductive” model for the thermal problem. This is the appropriate limit for the thermal problem since the thermal conductivities of the various components are more comparable. Here you solve a 2D problem in `r` and `theta` with an anisotropic thermal conductivity.

rtimms, 2021-11-23:
Plan:

- [ ] Add cylindrical finite volume method and axisymmetric geometry 
- [ ] Add the simple resistor model from the paper as an example
- [ ] Add "two potential" model (this is a problem in cell radius, r, for the current collector potentials
- [ ] Add axisymmetric thermal model
- [ ] Allow theta dependence (update geometry, etc.)
- [ ] Add 2D thermal model (coupling into the "two potential" model is through the theta-averaged temperature)

Yanisiqi, 2024-12-29:
I notice that there is one notebook named Jelly roll model: https://docs.pybamm.org/en/stable/source/examples/notebooks/models/jelly-roll-model.html
It seems that this model has not been integrated into PyBaMM yet: "Such functionality will be added to PyBaMM in a future release and will enable efficient simulations of jelly roll cells."
May I ask when do you plan to release it? Cylindrical cell is quite common in consumer electronics and I hope this functionality may be added in the future.