pybamm-team/PyBaMM #1318: Casadi event handling

rtimms, 2021-01-04:
Explore some other options for how events are handled by the casadi solver 

This is motivated by an issue @DrSOKane is having where the cut off voltage isn’t being fixed:

[image]

I think this is related to having to do “step-and-check” integration to handle events in casadi. The solver returns the times in t_eval (which is controlled by the period). When an event is crossed we create a linear interpolant, then use a root finding algorithm to find t_event. We then go back and solve up to that time to get y_event. Obviously when the solution is highly nonlinear the t_event we found isn’t correct and therefore the event state is out too. One solution would be to reduce the period, but then you end up with more data. We could also try using a higher order intepolant on line 229 of casadi_solver.py. This kind of event checking happens internally for the other solvers, so they can use the internal adaptive steps taken by the solver rather than the times in t_eval, and therefore hit the event more accurately. Ideally we would be able to access the internal steps used by the casadi solver for use in the root finding. 

An alternative would be to do some iteration in the event location part of the solver. I.e. find t_event using interpolation, solve up to this time, check again if the event has been crossed. If it has, use root finding again. If it hasn’t, solve a little further. Repeat. 

Another option would be the following. When an event is crossed, go back and integrate over that window of time again requesting the solution at a more dense set of points before doing the root finding.

tinosulzer, 2021-01-04:
I like the iterative solution. Hopefully converges in 2-3 iterations since Newton-Raphson has quadratic convergence?

Another similar option is to directly pass the solver to the rootfinding algorithm, so the rootfinder itself is calling the solver. But I don't know whether this would work since we are looking for one event to be zero, not trying to find the point where they are all zero

rtimms, 2021-01-04:
Yeah it will be interesting to see what the trade off is in doing multiple integrations in an iteration loop vs just doing one dense solve to get a solution you trust everywhere in that window of time (to some tol) and using that to find the event time. The problem with the latter is how many points is enough?

I agree iteration should converge quickly.

rtimms, 2021-01-08:
The issue seems to be related to the period, e.g.

```python3
import matplotlib.pyplot as plt
import pybamm

exp = pybamm.Experiment(
    ["Hold at 4.2V until C/100", "Rest for 4 hours (3 minute period)"]
    + ["Discharge at 1C until 2.5V (60 second period)"]
)

options = {"sei": "solvent-diffusion limited"}
model = pybamm.lithium_ion.DFN(options=options)

params = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Chen2020)
params.update({"Reference temperature [K]": 278.15})
params.update({"Ambient temperature [K]": 278.15})
params.update({"Initial temperature [K]": 278.15})
params.update({"Upper voltage cut-off [V]": 4.21})

periods = [10, 60]
markers = ["x", "o"]
for period, marker in zip(periods, markers):
    exp = pybamm.Experiment(
        ["Hold at 4.2V until C/100", "Rest for 4 hours (3 minute period)"]
        + [f"Discharge at 1C until 2.5V ({period} second period)"]
    )
    sim = pybamm.Simulation(model=model, parameter_values=params, experiment=exp)
    sim.solve()
    time = sim.solution["Time [s]"].entries
    voltage = sim.solution["Terminal voltage [V]"].entries
    plt.plot(
        time, voltage, linestyle="-", marker=marker, label=f"({period} second period"
    )
plt.legend()
plt.show()
```

[image]

rtimms, 2021-03-29:
Closing this now that #1450 is merged. We can make a new issue in the future if we have a specific event handling method we want to try.

closed 2021-03-29 by rtimms