pybamm-team/PyBaMM #1330: Enforce `print_citations()` in notebooks

tinosulzer, 2021-01-20:
All notebooks should have the cell `pybamm.print_citations()` at the end, to encourage people to use this command.
This can be enforced using regex when running the example tests.

brosaplanella, 2021-01-21:
I agree. I would also suggest then to print citations in a more friendly format (probably a LaTeX format won't be possible) but at least some sort of plain text reference list. This links to issue #1182.

rtimms, 2021-01-21:
pybtex seems to work ok now

e.g. 

```python3
import pybamm
import pybtex

model = pybamm.lithium_ion.SPM()

sim = pybamm.Simulation(model)
sim.solve([0, 3600])
pybamm.print_citations(filename="test.bib")
pybtex.format_from_file("test.bib", "unsrt", output_backend="plaintext")
```

you can output to markdown too and pick different styles

brosaplanella, 2021-01-21:
That is really cool! Is there a way of doing it without generating a bibtex file every time?

rtimms, 2021-01-22:
You can point to a particular file and then pick some keys to cite

```python3
print(pybtex.format_from_file("test.bib", "unsrt", citations=["sulzer2020python", "harris2020array"], output_backend="plaintext"))   
```

In theory we should just be able to point to our CITATIONS.txt and pass the relevant keys, but I was getting an error when I tried to do this. I think it is a bit picky about how things are typed out/formatted in the bib file...

rtimms, 2021-01-22:
Yeah, e.g doing 

```python3 
print(pybtex.format_from_file("pybamm/CITATIONS.txt", "unsrt", citations=["sulzer2020python", "harris2020array"], output_backend="plaintext"))   
``` 
works, but if you don't pass the citations argument (so that it print all citations) it fails. Think it will be a boring job of fixing commas etc, but then in future we can test to make sure it prints ok so that you just fix a single entry.

brosaplanella, 2021-01-22:
Sounds good! It seems that in order to print all the citations you need to pass `"[*]"`. I am happy to give it a go as it lines up with #1182  which I need to tackle at some point.

PR #1340 (2021-01-25): Enforce print_citations() in notebooks
# Description

- Notation in citations has been homogenised.
- `pybamm.print_citations()` has been extended to print formatted citations (both in screen and to files).
- Added examples on how to cite in Tutorials 1 and 2 (the comment in Tutorial 1 is a bit longer as it is the introduction).
- Test to check if notebooks include `pybamm.print_citations()` (commented out for the moment).

Fixes #1330 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the [... cut]
closed 2021-01-25 by brosaplanella