pybamm-team/PyBaMM #1362: Remove `Simplification` class and associated tests

tinosulzer, 2021-02-04:
Following #1301, the `Simplification` class is no longer required (see https://github.com/pybamm-team/PyBaMM/pull/1359#issuecomment-773442388)

The following can be removed:
- everything in `simplify.py` **except** `simplify_in_constant` (which could be moved to `symbol.py`)
- all simplifications in the `BaseSolver` class
- the `use_simplify` attribute of a model
- all `def simplify` function methods (add a clear error message to the one in `Symbol`)
- anywhere `.simplify()` appears
- testing simplifications in the `OptimisationsTest`
- all other associated tests and docs

I don't think anyone is using this so we are good to just delete instead of deprecating, imo

This is a bit of a stretch for a "good first issue" but should be doable for someone new unless the tests break in an unexpected way

asinghgaba, 2021-02-04:
Hi @tinosulzer , I'm new to open source, but think I can do these tasks. I understood all of them but had a couple of questions before I could begin-

> all def simplify function methods (add a clear error message to the one in Symbol)
For this, I didn't understand what kind of error message you want(and where).

And secondly, is this urgent? I might be able to do this over the weekend.

Thank you.

tinosulzer, 2021-02-04:
Thanks @asinghgaba , that would be great! Really not urgent.

What I mean is replace this line https://github.com/pybamm-team/PyBaMM/blob/afb920a5957a904ba833c8483385c31de42bb4d9/pybamm/expression_tree/symbol.py#L791
with 
```python3
raise pybamm.ModelError(".simplify() has been deprecated as it now has no effect")
```
(then add a test).

That is actually the only `def simplify` you should edit. The one in `Simplification` will be removed when you remove the whole class, and leave the ones in `Settings`

asinghgaba, 2021-02-04:
Okay @tinosulzer , sounds good. I'll start with it then.

PR #1369 (2021-02-08): Remove 'simplification' and other associated tests #1362
# Description

Removed `Simplification` class and relevant tests associated with it. Also, moved `simplify_if_constant()` to `symbol.py`.
Fixes #1362 
Completed all the tasks specified in #1362 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimiz [... cut]
closed 2021-02-08 by tinosulzer