pybamm-team/PyBaMM #1363: Incorrect values for y axis in 2D simulation?

DavidMStraub, 2021-02-05:
**Describe the bug**

This might not be a bug but just me not understanding how to do things properly (PyBaMM beginner...).
When using a 2+1D pouch cell model, the plot uses values for the y (horizontal) axis which are not what I would expect based on the `Electrode width` parameter setting.

**To Reproduce**
Steps to reproduce the behaviour:

```python
import pybamm

options = {"cell geometry": "pouch", "thermal": "x-lumped", "dimensionality": 2}
model = pybamm.lithium_ion.SPM(options=options)
geometry = model.default_geometry

parameter_values = model.default_parameter_values
parameter_values.update(
    {
        "Electrode height [m]": 0.3,
        "Electrode width [m]": 0.1,
        "Negative tab width [m]": 0.03,
        "Positive tab width [m]": 0.03,
        "Negative tab centre y-coordinate [m]": 0.03,
        "Positive tab centre y-coordinate [m]": 0.07,
        "Negative tab centre z-coordinate [m]": 0.3,
        "Positive tab centre z-coordinate [m]": 0.3,
    }
)
parameter_values.process_model(model)
parameter_values.process_geometry(geometry)

sim = pybamm.Simulation(model, parameter_values=parameter_values, solver=pybamm.CasadiSolver("fast"))
sim.solve([0, 60])
```
**Expected behaviour**

I expected the horizontal axis (y) to extend from 0 to 10 cm.

**Screenshots**

[image]

**Additional context**

Using PyBaMM 0.3.0 or current master (no difference). Python 3.8.5.

tinosulzer, 2021-02-05:
Yeah that's a bug, thanks for spotting. Fixing now

tinosulzer, 2021-02-05:
@all-contributors add @DavidMStraub for bug

tinosulzer, 2021-02-05:
FYI, you can pass the plot function a "spatial_unit" argument to display mm or m instead of um https://pybamm.readthedocs.io/en/latest/source/plotting/quick_plot.html

closed 2021-02-05 by tinosulzer

DavidMStraub, 2021-02-06:
Thanks for the quick fix!

PR #1364 (2021-02-05): Issue 1363 length scale bug
# Description

Changes the "current collector y" length scale from `L_y` to `L_z`. This is because both the y- and z-directions should be scaled with `L_z` when non-dimensionalising (otherwise there is an awkward extra factor of `L_y/L_z` in the 2D Laplacian for the current collector problem).

Fixes #1363 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking cha [... cut]