pybamm-team/PyBaMM #1373: Broken links

rtimms, 2021-02-10:
There are some broken links that don't seem to be getting picked up by the link checker. Two I have spotted are the Read The Docs in the README
[image]

and one of the links in `change-settings.ipynb`
[image]

@priyanshuone6 do you know why these get missed?

priyanshuone6, 2021-02-10:
I checked and found that the link checker is not picking up these links due to which they are not getting checked. I am not sure why this happening. I will try to find other options to maybe replace this workflow.

rtimms, 2021-02-10:
great, thanks!

tinosulzer, 2021-02-10:
maybe it doesn't check things that don't start with `https`?

priyanshuone6, 2021-02-11:
`[Read The Docs](readthedocs.io)` and `[here](../simulation-class.ipynb)` are without `https`.
I think that's why the URL checker is not picking up these links.

PR #1431 (2021-03-15): fix documentation and links
# Description

Fix documentation and links
Also fixes #1373 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ] [... cut]
closed 2021-03-15 by tinosulzer