pybamm-team/PyBaMM #1376: PEP fix styling around docstring

priyanshuone6, 2021-02-11:
I found these styling errors around docstring in functions in various files that doesn't adhere to Python's [PEP guidelines](https://www.python.org/dev/peps/pep-0257/#specification).
https://github.com/pybamm-team/PyBaMM/blob/9883106d84a8ad43b299ea4e0ce8ef58d395cb28/pybamm/citations.py#L29-L30
https://github.com/pybamm-team/PyBaMM/blob/9883106d84a8ad43b299ea4e0ce8ef58d395cb28/pybamm/settings.py#L32-L33
https://github.com/pybamm-team/PyBaMM/blob/9883106d84a8ad43b299ea4e0ce8ef58d395cb28/pybamm/util.py#L63-L64

**Expected behavior**
Find all those functions and replace `"` with 3 double quotes `"""` for consistency across the repository .

It can be marked as a good first issue

asinghgaba, 2021-02-11:
If anyone’s not working on it, I can get this done.

tinosulzer, 2021-02-12:
There are probably several other places where it's like this, I wasn't aware of the PEP guideline for this

PR #1383 (2021-02-12): #1376 Fixed PEP styling around docstrings
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #1376 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization  [... cut]
closed 2021-02-12 by tinosulzer