pybamm-team/PyBaMM #1377: Local ECM resistance

DavidMStraub, 2021-02-11:
**Describe the bug**

Hi, I am trying to understand the "Local ECM resistance" variable. It seems like this is [defined](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/base_battery_model.py#L904-L934) as the sum of all overpotentials, plus the difference between current OCV and initial OCV, divided by the current. If I understand correctly, this corresponds to treating the entire cell as a single Ohmic resistor with a time-dependent resistance. However, what I would have expected is that "ECM" refers to a model with a voltage source for OCV in series with an Ohmic resistor, so the OCV is not considered in the voltage difference.

It could be that this is intentional, so I would just like to clarify.

**To Reproduce**

```python
import pybamm
model = pybamm.lithium_ion.SPM()
sim = pybamm.Simulation(model)
sim.solve([0, 3600])
sim.plot(["Local ECM resistance [Ohm]"])
```

**Expected behaviour**

I would have expected the internal resistance (defined as the resistance in series with the OCV voltage source) to not vary drastically during discharge.

**Screenshots**

[image]

**Additional context**
Reproduced under v0.3.0 and on master (903b4a05ef5a4f91633e990d4aec12c53df723a2).

tinosulzer, 2021-02-11:
Hmm yeah I agree it probably shouldn't include the OCV change - @TomTranter ?

TomTranter, 2021-02-11:
Yeah it was intentional but change in ocv is also another variable so could
be taken out of this one

tinosulzer, 2021-02-11:
Yeah I think not having it in this variable is more intuitive. @DavidMStraub could you open a PR?

PR #1380 (2021-02-12): Remove OCV from ECM resistance definition
# Description

To address #1377, this removes the OCV (difference between current OCV and initial OCV) from the definition of the `Local ECM resistance`. The `Change in measured open circuit voltage` variable is left untouched.

Fixes #1377

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tes [... cut]
closed 2021-02-12 by tinosulzer