pybamm-team/PyBaMM #1386: Problem with parameter_sets.py auto-generated docs

tinosulzer, 2021-02-15:
- The automatic PR should only be opened when there is a push to develop (not any branch as is currently the case)
- Current formatting is making the doctests fail

priyanshuone6, 2021-02-15:
Could you please elaborate more on the doctest failing?

tinosulzer, 2021-02-15:
Here's the broken test: https://github.com/pybamm-team/PyBaMM/runs/1889607759#step:11:207
You can test this locally with `python run-tests.py --doctest`

priyanshuone6, 2021-02-16:
I tried to solve the error `/home/priyanshu/work/PyBaMM/pybamm/parameters/parameter_sets.py:docstring of pybamm.parameters.parameter_sets:9:Unexpected indentation.
FAILED`  but I am unable to find the problem in line 9 which is causing the tests to fail. Could you please help me with this?

tinosulzer, 2021-02-16:
rst is quite picky about indentations in lists. You could try isolating the problem by commenting things out and running the local test. I think sometimes they want a `\` at the end of the line in a list

PR #1387 (2021-02-16): Fix workflow and update_parameter_sets.py
# Description

Add 'develop' branch in the workflow
Add extra indentation in docstring
Fixes #1386
closed 2021-02-16 by tinosulzer