pybamm-team/PyBaMM #1388: Operating mode docs

rtimms, 2021-02-17:
The “operating mode” option is missing from the doc string for `BaseBatteryModel`

Saransh-cpp, 2021-02-25:
Can you please elaborate on what should be added to description of "operating mode" ? Should I add this - "Sets the operating mode for the model. Can be "Current" (default), "voltage" or "power""? There was also an ```elif``` condition while checking for the "operating mode" -
```python
elif callable(self.options["operating mode"]):
            self.submodels[
                "external circuit"
            ] = pybamm.external_circuit.FunctionControl(
                self.param, self.options["operating mode"]
            )
```
What can I add for this in the docstring? Thanks!

rtimms, 2021-02-26:
You could say something like "Alternatively, the operating mode can be controlled with an arbitrary function by passing the function directly as the option. In this case the function must define the residual of an algebraic equation. The applied current will be solved for such that the algebraic constraint is satisfied."

Saransh-cpp, 2021-02-26:
Thanks! I will add this in the doc string.

PR #1404 (2021-02-26): Added 'operating mode' to the doc string of BaseBatteryModel class
# Description

1. Added description for 'operating mode' parameter in BaseBatteryModel class.
2. Deleted '\\' from some lines to format the displayed output. With '\\', the output was behaving weirdly as displayed in the screenshot below (The line, where "average"  is described, was breaking up with huge spaces in between).
[image]

Please let me know if you want any changes or additions in the PR. Thanks!

Fixes #1388 

## Type of change

Doc string addition

- [ ] New feature (non-breaking change which adds functionality)
- [ [... cut]
closed 2021-02-26 by tinosulzer