pybamm-team/PyBaMM #1389: Using pyBaMM for modeling supercapacitors

salazardetroya, 2021-02-18:
I am interested in modeling a supercapacitor whose cyclic voltammetry is as follows:

[image]

I am able to capture the overall shape with a simple heat equation and effective properties through Bruggeman, but the bottom slanted line remains elusive. I have been told that it could be due to side reactions or leakage resistance. Can pyBaMM be applied for this type of models? Apologies if this is not the right place for user questions.

tinosulzer, 2021-02-18:
I am not hugely familiar with cyclic voltammetry but it sounds doable. How do you solve the model (or perform the experiment) to generate the above plot? Just a triangular current profile?

salazardetroya, 2021-02-18:
Sorry I should have added details on that. A cyclic voltammetry is simple an applied voltage that increases/decreases with time until a certain threshold/zero voltage is reached and then repeat. So yes, a triangular current profile.

[image]

Any ideas of what do add to the modeling equations to capture the bottom slanted line? My current modeling equation is

<img src="https://render.githubusercontent.com/render/math?math=\kappa \nabla^{2} \phi=a C_{D L} \frac{\partial \phi}{\partial t}">

where $\kappa$ is the conductivity in the electrode, $\phi$ is the potential and $a C_{D L}$ is the capacitance times specific area.

tinosulzer, 2021-02-19:
Would be great to have a special `Experiment` called `CyclicVoltammetry` which automatically solves the model with this voltage profile. Then you can test out the different models in pybamm (which are all a fair bit more complex than the simple parabolic equation you have) to see if any give you the slanted line. I don't have enough intuition with these plots to know what the effect is off the top of my head

brosaplanella, 2021-02-19:
I am not super familiar with supercapacitors, but aren't they sometimes modelled similarly to batteries, just with an additional capacitance term which, for supercaps, is dominant?

martinjrobins, 2021-02-20:
I use pybamm to model cyclic voltammetry and other similar input voltage waveforms (like Fourier transformed large amplitude AC voltammetry). A special `Experiment` class for cyclic voltammetry would be useful. For the moment I just write the equation directly, in which case specifying the voltage input term is straightforward, for example:

```python
Edc_forward = -pybamm.t
Edc_backwards = pybamm.t - 2*t_reverse
Eapp = E_start + \
    (pybamm.t <= t_reverse) * Edc_forward + \
    (pybamm.t > t_reverse) * Edc_backwards
```