pybamm-team/PyBaMM #1399: Polynomial particle concentration wrong in 2+1D models

rtimms, 2021-02-25:
If you use any of the polynomial particle concentration models with "dimensionality" equal to 2 you get incorrect behaviour (see example below). The example works as expected with "dimensionality" equal to 1 (and in standard 1D models). My first guess is that this could related to averaging somewhere (e.g. as in #1396), but I haven't investigated properly. 

```python3
import pybamm

models = [
    pybamm.lithium_ion.SPMe(name="1D Fickian"),
    pybamm.lithium_ion.SPMe({"particle": "uniform profile"}, name="1D uniform"),
    pybamm.lithium_ion.SPMe(
        {"current collector": "potential pair", "dimensionality": 2},
        name="2+1D Fickian",
    ),
    pybamm.lithium_ion.SPMe(
        {
            "current collector": "potential pair",
            "dimensionality": 2,
            "particle": "uniform profile",
        },
        name="2+1D uniform",
    ),
]
var = pybamm.standard_spatial_vars
var_pts = {
    var.x_n: 5,
    var.x_s: 5,
    var.x_p: 5,
    var.r_n: 5,
    var.r_p: 5,
    var.y: 5,
    var.z: 5,
}
sols = []
for model in models:
    sim = pybamm.Simulation(model, var_pts=var_pts)
    sol = sim.solve([0, 250])
    sols.append(sol)

pybamm.dynamic_plot(
    sols,
    [
        "Minimum positive particle surface concentration",
        "Maximum positive particle surface concentration",
        "Terminal voltage [V]",
    ],
)
```

[image]

rtimms, 2021-02-25:
Note for these parameters you expect the 1D and 2+1D plots plots here to be basically the same (very high current collector conductivity so basically uniform behaviour).

tinosulzer, 2021-02-26:
This could be linked to #1396 (`yz_average` doesn't do what you expect)?

rtimms, 2021-02-26:
Yeah I’m pretty sure that will be what is happening. Don’t notice in 1+1D since `l_z` is 1.

rtimms, 2021-03-02:
Just to update, this works ok if the model is DFN, so the problem results from using `PolynomialSingleParticle`. Digging now...

rtimms, 2021-03-02:
OK, so this was actually due to how the equation was being discretised by the finite element method. We need to write the source terms on the rhs using `pybamm.source(rhs, var)` so that they get multiplied by the mass matrix when they are discretised. We should set it up to handle this automatically, but I'll push the fix for now and do automation in a separate issue.

rtimms, 2021-03-03:
Fixed by #1411

closed 2021-03-03 by rtimms