pybamm-team/PyBaMM #1403: Import Error on MacOS Catalina

thuwangming, 2021-02-26:
Hello!
I have installed python@3.8, python@3.9 and sundials via Homebrew, and PyBaMM, casadi solver via pip on my Macbook. But 
 when I import pybamm module, it raised errors either on python@3.8 or python@3.9.
When using python@3.8, the error logs are as follows:
`File "<stdin>", line 1, in <module>
  File "/Users/wangming/Library/Python/3.8/lib/python/site-packages/pybamm/__init__.py", line 231, in <module>
    from .solvers.idaklu_solver import IDAKLUSolver, have_idaklu
  File "/Users/wangming/Library/Python/3.8/lib/python/site-packages/pybamm/solvers/idaklu_solver.py", line 13, in <module>
    idaklu = importlib.util.module_from_spec(idaklu_spec)
ImportError: dlopen(/Users/wangming/Library/Python/3.8/lib/python/site-packages/idaklu.cpython-38-darwin.so, 2): Library not loaded: /usr/local/opt/sundials/lib/libsundials_sunlinsolklu.3.5.0.dylib
  Referenced from: /Users/wangming/Library/Python/3.8/lib/python/site-packages/idaklu.cpython-38-darwin.so
  Reason: image not found`

However, it gave another error when using python@3.9:

`File "/Users/wangming/Downloads/Codes/battery/first.py", line 3, in <module>
    import pybamm
  File "/Users/wangming/Library/Python/3.9/lib/python/site-packages/pybamm/__init__.py", line 83, in <module>
    from .expression_tree.interpolant import Interpolant
  File "/Users/wangming/Library/Python/3.9/lib/python/site-packages/pybamm/expression_tree/interpolant.py", line 5, in <module>
    from scipy import interpolate
  File "/usr/local/lib/python3.9/site-packages/scipy/interpolate/__init__.py", line 165, in <module>
    from .interpolate import *
  File "/usr/local/lib/python3.9/site-packages/scipy/interpolate/interpolate.py", line 13, in <module>
    import scipy.special as spec
  File "/usr/local/lib/python3.9/site-packages/scipy/special/__init__.py", line 636, in <module>
    from . import _basic
  File "/usr/local/lib/python3.9/site-packages/scipy/special/_basic.py", line 16, in <module>
    from . import orthogonal
  File "/usr/local/lib/python3.9/site-packages/scipy/special/orthogonal.py", line 81, in <module>
    from scipy import linalg
  File "/usr/local/lib/python3.9/site-packages/scipy/linalg/__init__.py", line 194, in <module>
    from .misc import *
  File "/usr/local/lib/python3.9/site-packages/scipy/linalg/misc.py", line 4, in <module>
    from .lapack import get_lapack_funcs
  File "/usr/local/lib/python3.9/site-packages/scipy/linalg/lapack.py", line 783, in <module>
    from scipy.linalg import _flapack
ImportError: dlopen(/usr/local/lib/python3.9/site-packages/scipy/linalg/_flapack.cpython-39-darwin.so, 2): Symbol not found: _cgemqrt_
  Referenced from: /usr/local/lib/python3.9/site-packages/scipy/linalg/_flapack.cpython-39-darwin.so
  Expected in: flat namespace
 in /usr/local/lib/python3.9/site-packages/scipy/linalg/_flapack.cpython-39-darwin.so`

And everything works fine on Windows 10.
How can I solve this problem?
Thanks a lot!

Ming WANG

tinosulzer, 2021-02-26:
For the first error we are working on fixing it, there are some PRs open.
For the second error I haven't seen that before, could you try updating your scipy to the most recent version?

thuwangming, 2021-02-27:
> For the first error we are working on fixing it, there are some PRs open.
> For the second error I haven't seen that before, could you try updating your scipy to the most recent version?

I have updated scipy to the latest version, and it works. Thank you for your help!

tinosulzer, 2021-02-27:
Great! Closing as we already know about the first error :)

closed 2021-02-27 by tinosulzer