pybamm-team/PyBaMM #1426: readthedocs.io is still using the master branch to show docs

priyanshuone6, 2021-03-12:
**Describe the bug**
While reading the PyBaMM documentation from [pybamm.readthedocs.io](https://pybamm.readthedocs.io/en/latest/), I noticed that in [Adding a Model](https://pybamm.readthedocs.io/en/latest/tutorials/add-model.html) various links like the `create a model`, `expression tress` are still broken, because [pybamm.readthedocs.io](https://pybamm.readthedocs.io/en/latest/) is still using the master branch to show its docs, as all the broken links were already fixed in the develop branch using the link checker workflow.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to https://pybamm.readthedocs.io/en/latest/tutorials/add-model.html
2. Click on `create a model` or `expression tress`
[image]

**Expected behaviour**
[pybamm.readthedocs.io](https://pybamm.readthedocs.io/en/latest/) shows the latest and updated files from the develop branch instead of the master branch.

tinosulzer, 2021-03-12:
Thanks for catching this @priyanshuone6 . We already had the webhook, I updated it to point to develop. Your test works now

tinosulzer, 2021-03-12:
@all-contributors add @priyanshuone6 for bugs

closed 2021-03-12 by tinosulzer