pybamm-team/PyBaMM #1433: Composite anode of graphite and silicon

weilongai, 2021-03-16:
**Summary**
There is a trend of adding silicon to graphite to increase the energy density of batteries, so we are considering to develop a composite anode model. It is based on the fickan's diffusion equation for the two phases, but PyBaMM considers only one phase. What are the best ways to add this feature?

It seems not possible to reuse the particle submodel for lithium diffusion in both phases. Do we need to rewrite the particle submodel to make one particle_composite model?

I cannot explain the theoretical model too much, because it is under development.

tinosulzer, 2021-03-16:
One option is to add the particle_composite submodel, another option (more general) is to make the existing particle submodel more general and then call it twice, once for silicon and once for graphite. For example, see how the [base interface submodel](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/interface/base_interface.py) takes a `reaction` argument.

We will also need to think about how to deal with the two interfacial current densities for silicon and graphite, but this shouldn't be too bad as it's similar to SEI.

Either way I recommend you start by editing the `BasicDFN` model where you can easily just add another `Variable` for silicon concentration and edit the current equations

weilongai, 2021-03-16:
Thanks for your suggestions. I think I will make a new `particle_composite` submodel, because reusing the existing particle submodel needs a lot of modifications and is not an easy task.

weilongai, 2021-03-25:
Is there any example on how to use the `BasicDFN` model? I tried to use it but got the error of `KeyError: "'Total current density' not found` using the below codes.

```python3
import pybamm
import os
import numpy as np
import matplotlib.pyplot as plt
import timeit
start = timeit.default_timer()
os.chdir(pybamm.__path__[0]+'/..')

model = pybamm.lithium_ion.BasicDFNComposite()
chemistry = pybamm.parameter_sets.Chen2020_composite
param = pybamm.ParameterValues(chemistry=chemistry)
param.update({"Upper voltage cut-off [V]": 4.21})
param.update({"Lower voltage cut-off [V]": 2.49})

exp = pybamm.Experiment(["Discharge at 1C until 2.5V"], use_simulation_setup_type='old')
sim1 = pybamm.Simulation(
    model, 
    experiment=exp,
    parameter_values=param,
    solver=pybamm.CasadiSolver(),
)
solution1 = sim1.solve()

stop = timeit.default_timer()
print('running time: ' + str(stop - start) +'s')
```

tinosulzer, 2021-03-25:
It doesn't work with the Experiment. You have to run it like a regular simulation (see e.g. the "compare_lithium_ion" example)

weilongai, 2021-03-25:
Thanks, I have read `compare_lithium_ion.py`, but it is quite simple with nearly no options, for functions of applied currents, voltage windows and thermal model. I found one example for the `DFN` model in `compare-ecker-data.ipynb`, which I guess is used the same way as `basicDFN`, right?

Below is a copy of the `compare_lithium_ion.py`.
```python
#
# Compare lithium-ion battery models
#
import pybamm

pybamm.set_logging_level("INFO")

# load models
models = [pybamm.lithium_ion.SPM(), pybamm.lithium_ion.SPMe(), pybamm.lithium_ion.DFN()]

# create and run simulations
sims = []
for model in models:
    sim = pybamm.Simulation(model)
    sim.solve([0, 3600])
    sims.append(sim)

# plot
pybamm.dynamic_plot(sims)
```

tinosulzer, 2021-03-25:
You could also look at the `DFN.py` example. This set of examples takes you through how to set options in general https://github.com/pybamm-team/PyBaMM/tree/develop/examples/notebooks/Getting%20Started

tinosulzer, 2021-05-04:
How's this going @weilongai ? Am getting interested in silicon-graphite too :)

weilongai, 2021-05-05:
Hi @tinosulzer, I have finished the model in `basicDFN`, which seems to be enough for testing different parameters of silicon-graphite. A composite particle submodel is planned after we finish this work, which is more general for other kinds of composites, and I will need your help then. 

BTW, what studies are you going to do on silicon additives?

tinosulzer, 2021-05-05:
For now just understanding / modeling the OCV hysteresis

weilongai, 2022-03-15:
We have developed a basic composite electrode model for composite electrodes with two phases. e.g. graphite/silicon for the LG M50. Follow this link to get the code [
github.com/weilongai/PyBaMM/tree/composite-release](https://github.com/weilongai/PyBaMM/tree/composite-release)

This work is referred to a recent paper below
W. Ai, N. Kirkaldy, Y. Jiang, G. Offer, H. Wang, B. Wu (2022). A composite electrode model for lithium-ion battery with a silicon/graphite negative electrode. Journal of Power Sources. 527, 231142.

PR #2256 (2022-08-25): Composite release
# Description

Adds silicon/graphite composite model, using the two-phase changes from #2133 
Adds current sigmoid OCP submodel

Fixes #1433 
Fixes #2052 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking ch [... cut]
closed 2022-08-25 by tinosulzer