pybamm-team/PyBaMM #1445: Individual periods for experiment steps

bessman, 2021-03-25:
**Summary**
 The `period` argument of `pybamm.Experiment` controls the frequency with which data is recorded. This argument is currently a single value, applied to every step of the experiment. It would be useful to be able to set different periods for individual experiment steps.

 **Motivation**
When running an experiment where the current-step response is of interest, e.g. GITT or HPPC, it is necessary to set `period` to a very small value (<< 1 s) in order to capture transient effects close to the current step. Such a small period is not warranted further away from the current step, and only serves to inflate the size of the recorded data.

Using very small periods results in prohibitively large output data for longer experiments (multiple GB, all of which must be loaded into RAM when plotting).

**Additional context**
A comparison of a GITT pulse with `period="0.01 seconds"` and `period="1 minute"` (the default value).

[image]

In this example, it would be useful to be able to split the second and third steps into smaller steps with different periods. The periods would be smaller immediately after each current step, and get increasingly larger further away from the steps.

brosaplanella, 2021-03-25:
You can define custom periods for each step by adding `(30 second period)` at the end of your step statement. If you want different periods in the same step, you can define two identical steps with different period. For example:
```python3
experiment = pybamm.Experiment(
    (
        "Discharge at 2C until 2.5 V (30 second period)",
        "Rest for 15 minutes (30 second period)",
        "Rest for 105 minutes (5 minutes period)",
    )
)
```
does a 2 hour rest after discharge, but the first 15 minutes are sampled every 30 seconds to capture the transient effects, while the remaining 105 minutes are sampled at a much longer period. Does this help?

bessman, 2021-03-25:
Absolutely, that's exactly what I want to do. Thank you!

closed 2021-03-25 by bessman