pybamm-team/PyBaMM #1458: Parameterization example

tinosulzer, 2021-04-12:
Write a new example for parameterization showing:
- where the different parameter examples are
- how to find what are the parameters in a model
- how to add your own parameter set (in a dictionary)
- how to plot parameter functions

Saransh-cpp, 2021-04-12:
Could you please let me know if the things listed below are the ones that should be elaborated upon in the example notebook -
1. Different parameter examples
```bash
Setting parameter values.ipynb
parameter-management.ipynb
parameter-values.ipynb
```
2. Finding default parameters of a model
```py
model = pybamm.lithium_ion.DFN()    # Any model
default_parameters = model.default_parameter_values   
```
3. Adding parameter values (in a dictionary)
```py
def func(x):
       return x * 2
params = {"Some": 5, "Parameter": 10, "Values": 1, "A function": func}
parameter_values = pybamm.ParameterValues(params)
```
4. Parameter functions
```py
def current_func(time):
        return some_function

parameter_values["Current function [A]"] = current_func

sim = pybamm.Simulation(model, parameter_values=parameter_values)
sim.solve(t_eval)
sim.plot(output_variables)  # "Current [A]" in this case
```

Saransh-cpp, 2021-04-25:
@tinosulzer, any updates on this?

tinosulzer, 2021-04-25:
For 1 there is also the tutorial on read the docs which may need a bit of editing
For 2 I was thinking more the `model.parameters` attribute which shows what parameters actually appear in the expression tree. This function will also need to be edited to include `FunctionParameter` objects
For 4 I was thinking of showing how to do it without having the use the Simulation class (just call the function with the appropriate range and plot)

It would be good to show this in detail for a simple model, then also for the SPM and DFN

Thanks very much for taking this on @Saransh-cpp ! I thought I would have time to do it myself but easier if you do first and then I might tweak the example :)

Saransh-cpp, 2021-04-25:
> For 1 there is also the tutorial on read the docs which may need a bit of editing
> For 2 I was thinking more the `model.parameters` property which shows what parameters actually appear in the expression tree. This function will also need to be edited to include `FunctionParameter` objects
> For 4 I was thinking of showing how to do it without having the use the Simulation class (just call the function with the appropriate range and plot)
> 
> It would be good to show this in detail for a simple model, then also for the SPM and DFN
> 
> Thanks very much for taking this on @Saransh-cpp ! I thought I would have time to do it myself but easier if you do first and then I might tweak the example :)

I'll start working on this!

Saransh-cpp, 2021-04-27:
@tinosulzer should I use the model defined in [3-negative-particle-problem.ipynb](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/Creating%20Models/3-negative-particle-problem.ipynb) to explain the stuff? and then move to DFN and SPM?

The parameters will be obtained by `model.parameters` and the example already uses the "adding parameter values in a dictionary", then I can use `.update()` for further explanations

tinosulzer, 2021-04-27:
Yep that's a good one to use

rtimms, 2021-04-29:
@Saransh-cpp there are a couple of small issues related to parameters [here](https://github.com/pybamm-team/PyBaMM/discussions/1468) that might be worth fixing as part of this if it's not too much work?

The first is that when you do `params.export_csv(filename="export_param.csv")` you just get a csv with the parameter names and values. If you try to load that csv back as `pybamm.ParameterValues("export_param.csv")` you get an error as pybamm expects the columns to be labelled "Name [units]" and "Values". Could we get `export_csv` to automatically add the column headers?

The second is that if you try to update some parameters using an instance of `ParameterValues` e.g. 
```python3
model = pybamm.lithium_ion.BasicDFN()
values = model.default_parameter_values
anode_param = pybamm.ParameterValues("pybamm/input/parameters/lithium_ion/negative_electrodes/graphite_Chen2020/parameters.csv")
values.update(anode_param)
```
you get an error. The workaround is to just pass the dict items 
```python3
model = pybamm.lithium_ion.BasicDFN()
values = model.default_parameter_values
anode_param = pybamm.ParameterValues("pybamm/input/parameters/lithium_ion/negative_electrodes/graphite_Chen2020/parameters.csv")
values.update(anode_param._dict_items) 
```
. This isn't a huge deal tbh, but since `ParameterValues` is basically just a dictionary I wondered if it could be fixed. The problem seems to be [here](https://github.com/pybamm-team/PyBaMM/blob/b4735e19e4570dccc66aae90213e982547f4b532/pybamm/parameters/parameter_values.py#L337) when you do `"Typical current [A]" in values`.  

I can make a separate issue for these if you would prefer.

Saransh-cpp, 2021-04-29:
@rtimms I was going through the `csv` files present in `pybamm` and they have the headers as `Name [units],Value`, is only this required as the headers in the created csv files? I tried setting the `header` as -
```py
        df = pd.DataFrame(parameter_output)
        df = df.transpose()
        df.to_csv(filename, header=["Name [units],Value"])
``` 
it gives me this error while passing it in `ParameterValues`
```terminal
  File "c:\Users\Saransh\Saransh_softwares\Learning Projects\python\PyBamm\discussion.py", line 13, in <module>
    new_values = pybamm.ParameterValues("export_param.csv")
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pybamm\parameters\parameter_values.py", line 80, in __init__
    values = self.read_parameters_csv(file_path)
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pybamm\parameters\parameter_values.py", line 248, in read_parameters_csv
    return {k: v for (k, v) in zip(df["Name [units]"], df["Value"])}
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pandas\core\frame.py", line 3024, in __getitem__
    indexer = self.columns.get_loc(key)
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pandas\core\indexes\base.py", line 3082, in get_loc
    raise KeyError(key) from err
KeyError: 'Name [units]'
```

Saransh-cpp, 2021-04-29:
Update, this works - 
```py
        df = pd.DataFrame(parameter_output)
        df = df.transpose()
        df.to_csv(filename, header=['Value'], index_label="Name [units]")
```

Gives me an error - 
```terminal
Traceback (most recent call last):
  File "c:\Users\Saransh\Saransh_softwares\Learning Projects\python\PyBamm\discussion.py", line 13, in <module>
    new_values = pybamm.ParameterValues("export_param.csv")
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pybamm\parameters\parameter_values.py", line 84, in __init__
    self.update(values, check_already_exists=False, path=path)
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pybamm\parameters\parameter_values.py", line 318, in update
    data = pd.read_csv(
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pandas\io\parsers.py", line 605, in read_csv
    return _read(filepath_or_buffer, kwds)
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pandas\io\parsers.py", line 457, in _read
    parser = TextFileReader(filepath_or_buffer, **kwds)
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pandas\io\parsers.py", line 814, in __init__
    self._engine = self._make_engine(self.engine)
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pandas\io\parsers.py", line 1045, in _make_engine
    return mapping[engine](self.f, **self.options)  # type: ignore[call-arg]
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pandas\io\parsers.py", line 1862, in __init__
    self._open_handles(src, kwds)
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pandas\io\parsers.py", line 1357, in _open_handles
    self.handles = get_handle(
  File "C:\Users\Saransh\Saransh_softwares\Python_3.9\lib\site-packages\pandas\io\common.py", line 642, in get_handle
    handle = open(
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\Saransh\\Saransh_softwares\\Learning Projects\\python\\PyBamm\\graphite_LGM50_ocp_Chen2020.csv'
```

Note - `C:\\Users\\Saransh\\Saransh_softwares\\Learning Projects\\python\\PyBamm` is not the path of the actual pybamm repository so I don't think that this error is related to the headers.

rtimms, 2021-04-29:
Great, thanks!

PR #1471 (2021-05-13): Add parameterization example notebook
# Description

Added an example notebook explaining different concepts of parameterization using a simple model, and then with `DFN` and `SPM` models.

Overall the example is getting pretty big, please let me know if there are some steps that can be omitted or some steps that I missed and should be added.

View the notebook on my fork [here](https://github.com/Saransh-cpp/PyBaMM/blob/issue-1458-parameterization-example/examples/notebooks/parameterization.ipynb)

Fixes #1458 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](ht [... cut]
closed 2021-05-13 by tinosulzer