pybamm-team/PyBaMM #1465: Plot parameter functions

rtimms, 2021-04-26:
Some parameters are provided as functions (e.g. diffusivity as a function of concentration and temperature). Currently, to plot them you have to do something like this 
```python3
import pybamm

params = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Chen2020)
i0 = params["Negative electrode exchange-current density [A.m-2]"]
x = pybamm.linspace(3000,6000,100)
i0_eval = i0(1000,x,300)
i0_processed = params.process_symbol(i0_eval)
pybamm.plot(x, i0_processed)
```
We have to process the function "Negative electrode exchange-current density [A.m-2]" since its definition contains a `pybamm.Parameter`. It would be neat to be able to do something like `params.plot("Negative electrode exchange-current density [A.m-2]", (1000, x, 300)` instead so you can quickly plot the functional dependence of parameters. We can use `pybamm.plot` and `pybamm.plot2D` to plot the resulting arrays.

rtimms, 2021-04-26:
This is actually listed as one of the bullet points in #1458

Saransh-cpp, 2021-04-26:
> This is actually listed as one of the bullet points in #1458 .
> 
> @Saransh-cpp , just some thoughts here on plotting parameters

I'll have a look and see if I can implement the `plot` function in `ParameterValues`

tinosulzer, 2021-04-26:
Just a comment - when we provide custom plotting functions we need to be careful not to do so at the expense of matplotlib flexibility (specifying args and kwargs in the plot, adding xlabel and ylabel and legend, etc).

rtimms, 2021-04-27:
Good point re plotting. Maybe instead of adding extra plotting functionality, we could add something that makes it easier to get a callable function from `ParameterValues`? 

E.g. you can do 
```python3
import pybamm

def myfun(x):
    return pybamm.Parameter("D") * x

params = pybamm.ParameterValues({"D":2, "function":myfun})

def f(x):
    return params.process_symbol(params["function"](x))

x = pybamm.linspace(0,1,10)
pybamm.plot(x,f(x))
```

It would be handy to get `f(x)` directly without having to define it yourself. I couldn't figure out how to process the parameter and then call it, but maybe it doesn't matter too much. Also, it's not clear if we would want `f(x)` to return a pybamm array so we can use `pybamm.plot` or if it should just return a numpy array and people can do what they want with it.

It's not a huge effort to define `f(x)` yourself like this, but it's one extra step for new users to figure out.

Saransh-cpp, 2021-04-27:
This simple function does work but is, as it looks, limited to the above example (should be modified to add all plots in a single screen though). Should we generalize this or will it be better to just create a function for returning `params.process_symbol()`
```py
    def plot(self, values, eval):

        self.check_parameter_values(values)

        for value in values:
            processed = self.process_symbol(self[value](eval[0], eval[1], eval[2]))
            pybamm.plot(eval[1], processed, xlabel="time", ylabel=value)
```

rtimms, 2021-04-27:
Yeah, I think maybe it is better to just create a function that returns the processed parameter function. Each of the parameter functions may have a different number of arguments, and maybe trying to be clever with the plotting will get out of hand. 

Then `pybamm.plot` and `pybamm.plot2D` could be modified to take/return the axis (like `pybamm.plot_voltage_components`) to make it easier for people to customize the plots. 

thoughts @tinosulzer ?

tinosulzer, 2021-04-27:
Doesn't `params["function"]` already do exactly what you want (return the callable function object)?

> Then pybamm.plot and pybamm.plot2D could be modified to take/return the axis (like pybamm.plot_voltage_components) to make it easier for people to customize the plots.

agreed

rtimms, 2021-04-27:
Almost, but it returns an expression tree rather than an array. If the function itself contains a `Parameter` then you have to process the result to set the values, which feels a little awkward since `params` already knows the values. Basically I want a callable function that returns an array, in which all the `Parameter` symbols have been replaced by the values in `params`

tinosulzer, 2021-04-28:
I see the issue now. What about editing the `.evaluate()` function so it can return an array?

PR #1472 (2021-04-29): #1465 allow evaluate parameter to return arrays
# Description

`ParameterValues.evaluate()` can now return arrays so that you can evaluate parameter functions as 
```python3
import pybamm

params = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Chen2020)
i0 = params["Negative electrode exchange-current density [A.m-2]"]
x = pybamm.linspace(3000,6000,100)
params.evaluate(i0(1000,x,300))
```

Also, `plot` and `plot2D` now take and return the axis `ax` for easier customization
. 
Fixes #1465 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.co [... cut]
closed 2021-04-29 by rtimms