pybamm-team/PyBaMM #1469: Install Instructions

DrMichaelCornish, 2021-04-28:
The instructions for developer install found in the "Installation" section of https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md state that the user should run "tox -e pybamm-requires" in the second step (I'm on ubuntu 20.04). However, this creates an error. The step should specify that the user first clones the github repo and then change directory to the pybamm folder. Once that is done, the command "tox -e pybamm-requires" runs without issue. 

This order of operations is specified in the more detailed instructions linked to a the end of that installation section https://pybamm.readthedocs.io/en/latest/install/install-from-source.html

tinosulzer, 2021-04-28:
Thanks Michael, @tlestang  is it time to update the docs now with the new instructions for sundials?

tlestang, 2021-05-07:
Yes. For the specific issue of the developer install, I think the contributing guide should just point to the docs.

PR #1615 (2021-08-19): #1469 remove install instructions from CONTRIBUTIONG and point to ins…
…tall docs

# Description

See title

Fixes #1469 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs`  [... cut]
closed 2021-08-19 by martinjrobins