pybamm-team/PyBaMM #1470: import command typo

DrMichaelCornish, 2021-04-28:
In the file pybamm/solvers/jax_bdf_solver.py, the import command on line 12 is written 

from jax.tree_util import tree_map, tree_flatten, tree_unflatten, tree_multimap, partial

but the final term should be the capitalised "Partial" rather than "partial". As it is currently written, an error will prevent the python command "import pybamm" from successfully running.

Saransh-cpp, 2021-04-28:
This is something that I and @tinosulzer encountered too (specifically on ubuntu, with the develop branch), re-installing pybamm worked for us.

tinosulzer, 2021-04-29:
@DrMichaelCornish let us know if re-installing fixes this

raghuramshankar, 2021-09-06:
Hi,

I encountered the same issue. I tried to uninstall and reinstall pybamm, however this did not fix the issue for me. Please refer to the issue mentioned for my current installation.

Saransh-cpp, 2021-09-07:
Weird. Which version of pybamm are you using? Can you navigate to `pybamm/solvers/jax_bdf_solver.py` and check if `partial` is being imported from `jax.tree_util` or from `functools`?

raghuramshankar, 2021-09-07:
Hi,

Thank you for your reply.
Here is my pybamm installation:

```
pip show pybamm  
Name: pybamm
Version: 0.2.4.post3
```

I think I have just noticed another bug. Pip does not install the latest version of `pybamm` for me. I think I will create another issue for that.

In my installation, in `pybamm/solvers/jax_bdf_solver.py`, `partial` is imported from `jax.tree_util`.

Saransh-cpp, 2021-09-07:
Oh, then this might be a version issue. Updating pybamm would solve this.

raghuramshankar, 2021-09-07:
Trying to upgrade `pybamm` gives me the error:

```
pip install --upgrade pybamm==0.4.0 
Defaulting to user installation because normal site-packages is not writeable
Collecting pybamm==0.4.0
  Using cached pybamm-0.4.0-cp39-cp39-manylinux2010_x86_64.whl (1.5 MB)
Requirement already satisfied: jupyter in ./.local/lib/python3.9/site-packages (from pybamm==0.4.0) (1.0.0)
ERROR: Could not find a version that satisfies the requirement jaxlib==0.1.57 (from pybamm) (from versions: 0.1.60, 0.1.61, 0.1.62, 0.1.63, 0.1.64, 0.1.65, 0.1.66, 0.1.67, 0.1.68, 0.1.69, 0.1.70, 0.1.71)
ERROR: No matching distribution found for jaxlib==0.1.57
```

Maybe `pybamm` needs to be updated for the latest `jaxlib`.

tinosulzer, 2021-09-07:
Also it looks like you aren't using a virtual environment (based on the path `./.local/lib/python3.9/site-packages`). We strongly recomment using one (via virtualenv or conda) to avoid issues like this, compatibility clashes, etc. Not just for pybamm but all your python projects.

raghuramshankar, 2021-09-10:
Hi,

Thank you for the suggestion. I created a new virtual environment and tried `pip install pybamm==0.4.0.post1`. Here is the error I got:

```
pip install pybamm==0.4.0.post1
Collecting pybamm==0.4.0.post1
  Using cached pybamm-0.4.0.post1-cp39-cp39-manylinux2010_x86_64.whl (1.5 MB)
Collecting scipy>=1.3
  Using cached scipy-1.7.1-cp39-cp39-manylinux_2_5_x86_64.manylinux1_x86_64.whl (28.5 MB)
Collecting matplotlib>=2.0
  Using cached matplotlib-3.4.3-cp39-cp39-manylinux1_x86_64.whl (10.3 MB)
Collecting numpy>=1.16
  Using cached numpy-1.21.2-cp39-cp39-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (15.8 MB)
ERROR: Could not find a version that satisfies the requirement jaxlib==0.1.57 (from pybamm) (from versions: 0.1.60, 0.1.61, 0.1.62, 0.1.63, 0.1.64, 0.1.65, 0.1.66, 0.1.67, 0.1.68, 0.1.69, 0.1.70, 0.1.71)
ERROR: No matching distribution found for jaxlib==0.1.57
```

So if I understood this right, `pip install pybamm=0.4.0` requires `jaxlib==0.1.57`, which is the not available. Only `jaxlib>=0.1.60` is available. 

I wanted to check the version of `jaxlib` that `pybamm==0.2.4.post3` requires. So I created a new virtual environment and tried `pip install pybamm==0.2.4.post3`. Here is what I got:

```
pip install pybamm==0.2.4.post3
Successfully installed MarkupSafe-2.0.1 Send2Trash-1.8.0 anytree-2.8.0 argon2-cffi-21.1.0 attrs-21.2.0 autograd-1.3 backcall-0.2.0 bleach-4.1.0 casadi-3.5.5 cffi-1.14.6 cycler-0.10.0 debugpy-1.4.3 decorator-5.0.9 defusedxml-0.7.1 entrypoints-0.3 future-0.18.2 ipykernel-6.4.0 ipython-7.27.0 ipython-genutils-0.2.0 ipywidgets-7.6.4 jedi-0.18.0 jinja2-3.0.1 jsonschema-3.2.0 jupyter-1.0.0 jupyter-client-7.0.2 jupyter-console-6.4.0 jupyter-core-4.7.1 jupyterlab-pygments-0.1.2 jupyterlab-widgets-1.0.1 kiwisolver-1.3.2 matplotlib-3.4.3 matplotlib-inline-0.1.3 meshio-5.0.0 mistune-0.8.4 nbclient-0.5.4 nbconvert-6.1.0 nbformat-5.1.3 nest-asyncio-1.5.1 notebook-6.4.3 numpy-1.21.2 packaging-21.0 pandas-1.3.2 pandocfilters-1.4.3 parso-0.8.2 pexpect-4.8.0 pickleshare-0.7.5 pillow-8.3.2 prometheus-client-0.11.0 prompt-toolkit-3.0.20 ptyprocess-0.7.0 pybamm-0.2.4.post3 pycparser-2.20 pygments-2.10.0 pyparsing-2.4.7 pyrsistent-0.18.0 python-dateutil-2.8.2 pytz-2021.1 pyzmq-22.2.1 qtconsole-5.1.1 qtpy-1.11.0 scikit-fem-3.2.0 scipy-1.7.1 six-1.16.0 terminado-0.12.1 testpath-0.5.0 tornado-6.1 traitlets-5.1.0 wcwidth-0.2.5 webencodings-0.5.1 widgetsnbextension-3.5.1
```

It did not install `jax` nor `jaxlib`.

tinosulzer, 2021-09-13:
Yeah it looks like jax removes their old versions, which sucks. Sounds like you got this fixed though

closed 2021-09-13 by tinosulzer