pybamm-team/PyBaMM #1479: pybamm fails when running script a second time in spyder on windows

TomTranter, 2021-05-07:
**Describe the bug**
Running any script more than once seems to result in an import error because of the recent changes to parameters

  File "<frozen importlib._bootstrap>", line 908, in _find_spec

AttributeError: 'ImportDenier' object has no attribute 'find_spec'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "C:\Users\tom\code\PyBaMM\examples\scripts\drive_cycle.py", line 14, in <module>
    param = model.default_parameter_values

  File "c:\users\tom\code\pybamm\pybamm\models\full_battery_models\base_battery_model.py", line 346, in default_parameter_values
    return pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Marquis2019)

  File "c:\users\tom\code\pybamm\pybamm\parameters\parameter_values.py", line 73, in __init__
    self.update_from_chemistry(chemistry)

  File "c:\users\tom\code\pybamm\pybamm\parameters\parameter_values.py", line 216, in update_from_chemistry
    self.update(

  File "c:\users\tom\code\pybamm\pybamm\parameters\parameter_values.py", line 301, in update
    loaded_value = pybamm.load_function(os.path.join(path, value[10:]))

  File "c:\users\tom\code\pybamm\pybamm\util.py", line 273, in load_function
    module_object = importlib.import_module(path)

  File "C:\Users\tom\anaconda3\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)

  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import

  File "<frozen importlib._bootstrap>", line 991, in _find_and_load

  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked

  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed

  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import

  File "<frozen importlib._bootstrap>", line 991, in _find_and_load

  File "<frozen importlib._bootstrap>", line 971, in _find_and_load_unlocked

  File "<frozen importlib._bootstrap>", line 910, in _find_spec

  File "<frozen importlib._bootstrap>", line 884, in _find_spec_legacy

  File "C:\Users\tom\anaconda3\lib\site-packages\IPython\external\qt_loaders.py", line 48, in find_module
    if path:

  File "<frozen importlib._bootstrap_external>", line 1183, in __len__

  File "<frozen importlib._bootstrap_external>", line 1162, in _recalculate

  File "<frozen importlib._bootstrap_external>", line 1158, in _get_parent_path

KeyError: 'pybamm.input.parameters.lithium_ion'

tlestang, 2021-05-07:
Is this maybe related to #1475 ? @priyanshuone6 any ideas what is going wrong?

tlestang, 2021-05-07:
@TomTranter Could you provide a minimal example and steps to reproduce please?

priyanshuone6, 2021-05-07:
I could dig deeper after I get the steps to reproduce this bug

TomTranter, 2021-07-19:
I'm using spyder on windows - running any script twice without restarting the kernel results in the above error. Still happening - any ideas?

TomTranter, 2021-07-19:
Seems to be ok when I run example scripts from the pybamm directory but not scripts from another folder

TomTranter, 2021-07-19:
Think I fixed it by putting empty __init__ files into all the parameter subfolders

PR #1555 (2021-07-21): add empty inits to parameters subfolders
# Description

When running scripts through spyder on windows (at least this is what I tested) if you press run twice there is an importlib KeyError. The kernel had to be restarted to fix the issue. Adding empty __init__files to parameter subfolders fixes the issue

Fixes #1479

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New [... cut]
closed 2021-07-21 by TomTranter