pybamm-team/PyBaMM #1482: Discretization effects when simulating DFN model

faultdiagnosistoolbox, 2021-05-14:
Hi, 

I have a problem that looks like a simulation issue to me, but could also be that I've misunderstood something since I haven't worked with PyBaMM for long.

*Simulation*: Simulation of a DFN and SPM model with a sinusoid current load.
*Problem*: When simulating with high enough frequency (say, from f=2 kHZ) I see significant discretization effects in the simulated voltage for the DFN model, but not the SPM model.

Below is a minimal working example that illustrates the problem for me.

Regards, 
  Erik

```
import pybamm
import numpy as np
import matplotlib.pyplot as plt

model_dfn = pybamm.lithium_ion.DFN()
model_spm = pybamm.lithium_ion.SPM()

parameter_values = pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Chen2020)

# Current frequency
f = 20000  # Discretization effects
# f = 200  # Smooth solution
def my_current(t):
    return pybamm.sin(2 * np.pi * t * f)

parameter_values["Current function [A]"] = my_current
sim_dfn = pybamm.Simulation(model_dfn, parameter_values=parameter_values)
sim_spm = pybamm.Simulation(model_spm, parameter_values=parameter_values)

# 5 cycles
N = 2000  # Number of points in simulation output grid
t_eval = np.linspace(0, 1 / f * 5, N)

sim_dfn.solve(t_eval=t_eval)
sim_spm.solve(t_eval=t_eval)

sim_dfn.plot(["Current [A]", "Terminal voltage [V]"])
sim_spm.plot(["Current [A]", "Terminal voltage [V]"])

plt.show()
```

tinosulzer, 2021-05-14:
Thanks for pointing this out. Very strange indeed.
It seems to be a problem with the `CasadiSolver`, since changing the solver to `pybamm.ScikitsDaeSolver()` gives a smooth solution for the DFN

tinosulzer, 2021-05-14:
I'm trying to dig into this more - there might be a "minimum step" in the casadi solver or something.
If anyone is able to reproduce this in a MWE in casadi that we could ask them about that would be helpful

kratman, 2024-11-21:
This does not appear to be an issue anymore

closed 2024-11-21 by kratman