pybamm-team/PyBaMM #1483: Manipulate SEI resistance per unit thickness [Ohm.m-1]

gyouhoc, 2021-05-15:
Hi, I want to change the value of the SEI resistance per unit thickness [Ohm.m-1]. 

I add the seis in the parameter options and I view the list of the parameters but I don't see the sei related parameter in the list. Is there anything I miss or any mistake to pull up the sei parameters? 

parameter_options = {
    "chemistry": "lithium-ion",
    "cell": "nmc_peter2020",
    "anode": "graphite_peter2020",
    "separator": "separator_NMC2020",
    "cathode": "nmc_peter2020",
    "electrolyte": "LiPF6_peter2020",
    "experiment": "1C_discharge_from_full_peter2020",
    "seis":"Example"
}

parameter_values = pybamm.ParameterValues(chemistry=parameter_options)
parameter_values 

{'1 + dlnf/dlnc': 1.0,
 'Activation_energy_electrolyte': 34700,
 'Ambient temperature [K]': 298.15,
 'C1': -10.72,
 'C2': 23.88,
 'C3': -16.77,
 'C4': 2.595,
 'C5': 4.563,
 'Cation transference number': 0.2594,
 'Cell capacity [A.h]': 5.0,
 'Current function [A]': 5.0,
 'DESIGN_ANODE_DENSITY_HOST': 2266,
 'DESIGN_ANODE_MOLWT_HOST': 0.072064,
 'DESIGN_ANODE_NLI_HOST': 0.727,
 'DESIGN_ANODE_STOICHMAX': 1,
 'DESIGN_ANODE_VF_IM': 0.0351,
 'DESIGN_CATHODE_DENSITY_HOST': 4800,
 'DESIGN_CATHODE_MOLWT_HOST': 0.088945,
 'DESIGN_CATHODE_NLI_HOST': 0.75,
 'DESIGN_CATHODE_STOICHMAX': 1,
 'DESIGN_CATHODE_STOICHMIN': 0.384,
 'DESIGN_CATHODE_VF_IM': 0.125094,
 'DESIGN_CELL_ICL_FRAC_CATUSE': 0.02,
 'DESIGN_CELL_NP_RATIO': 1.152,
 'Edge heat transfer coefficient [W.m-2.K-1]': 0.3,
 'Electrode height [m]': 0.089,
 'Electrode width [m]': 2.34,
 'Electrolyte conductivity [S.m-1]': <function electrolyte_conductivity_Nyman2008 at 0x0000018755D01510>,
 'Electrolyte diffusivity [m2.s-1]': <function electrolyte_diffusivity_Nyman2008 at 0x0000018755D01378>,
 'Electrolyte_diffusion_activation_energy': 37040,
 'Initial concentration in electrolyte [mol.m-3]': 1200.0,
 'Initial concentration in negative electrode [mol.m-3]': 18081.0,
 'Initial concentration in positive electrode [mol.m-3]': 20090.0,
 'Initial temperature [K]': 298.15,
 'Lower voltage cut-off [V]': 2.5,
 'Maximum concentration in negative electrode [mol.m-3]': 25000.0,
 'Maximum concentration in positive electrode [mol.m-3]': 40500.0,
 'Negative current collector conductivity [S.m-1]': 58411000.0,
 'Negative current collector density [kg.m-3]': 7925.311203,
 'Negative current collector specific heat capacity [J.kg-1.K-1]': 385.0,
 'Negative current collector surface heat transfer coefficient [W.m-2.K-1]': 0.0,
 'Negative current collector thermal conductivity [W.m-1.K-1]': 401.0,
 'Negative current collector thickness [m]': 9.1e-06,
 'Negative electrode Bruggeman coefficient (electrode)': 1.5,
 'Negative electrode Bruggeman coefficient (electrolyte)': 1.5,
 'Negative electrode OCP [V]': ('graphite_LGM50_ocp_Chen2020',
                                array([[0.03129623, 1.0828807 ],
       [0.03499902, 0.99593794],
       [0.0387018 , 0.90023398],
       [0.04240458, 0.79649431],
       [0.04610736, 0.73354429],
       [0.04981015, 0.66664314],
[... 280 lines cut ...]
 'Positive tab centre z-coordinate [m]': 0.13699999999999998,
 'Positive tab heat transfer coefficient [W.m-2.K-1]': 10.0,
 'Positive tab width [m]': 2.35,
 'Reference OCP vs SHE in the negative electrode [V]': nan,
 'Reference OCP vs SHE in the positive electrode [V]': nan,
 'Reference temperature [K]': 298.15,
 'Separator Bruggeman coefficient (electrode)': 1.5,
 'Separator Bruggeman coefficient (electrolyte)': 1.5,
 'Separator density [kg.m-3]': 757.0,
 'Separator porosity': 0.46,
 'Separator specific heat capacity [J.kg-1.K-1]': 700.0,
 'Separator thermal conductivity [W.m-1.K-1]': 0.19,
 'Separator thickness [m]': 1.7e-05,
 'Typical current [A]': 5.0,
 'Typical electrolyte concentration [mol.m-3]': 1000.0,
 'Upper voltage cut-off [V]': 4.1,
 'anode_D_ref': 3.9e-14,
 'anode_E_D_s': 30000,
 'anode_E_ex': 35000,
 'anode_m_ref': 6.48e-07,
 'cathode_D_ref': 3e-14,
 'cathode_E_D_s': 30000,
 'cathode_E_ex': 50000,
 'cathode_m_ref': 3.59e-06,
 'cathode_stretch': 1}

tinosulzer, 2021-05-15:
That is weird, you should be seeing it. Can you try lower case "example" and also "ramadass2004" and "yang2017_sei" to see if any of those work?

gyouhoc, 2021-05-15:
I change "seis" to "sei" and it works. 
I think I have a typo.

closed 2021-05-15 by gyouhoc