pybamm-team/PyBaMM #1493: Infinite loop occurs

gyouhoc, 2021-05-25:
While I conduct multiple simulation with Pybamm, I find that the simulation generates an infinite loop and causes the python program to be halted. 

[image]

[image]

tinosulzer, 2021-05-25:
Can you try using `CasadiSolver("fast with events")` instead?

gyouhoc, 2021-05-25:
for me, ("fast with events") doesn't work. I tried three different solver options and "old safe" is working with an warning. 

"safe": SolverError: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=81.88051713742692).

"fast": SolverError: .../casadi/interfaces/sundials/idas_interface.cpp:569: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.

"old safe": working with an warning
            psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:827: Calculating Jacobian failed

tinosulzer, 2021-05-26:
Please provide an entire script that can be run to test this

gyouhoc, 2021-05-27:
Here is the script and required data.

tinosulzer, 2021-05-27:
I don't see it

closed 2021-07-17 by gyouhoc

Prasanna-murata, 2022-06-27:
Can you please say the solution to this in case you discussed it

Sunnyturicum, 2023-08-06:
Hey, I have the same issue. Is there a solution to this problem?

brosaplanella, 2023-08-07:
Please provide a minimum working example and the full error log, so we can look into it.

Sunnyturicum, 2023-08-07:
Thank you for taking the time. I have attached a minimal working example and full error log.
```
import pybamm
import matplotlib.pyplot as plt
import numpy as np

model = pybamm.lithium_ion.DFN(
    {
        "lithium plating": "partially reversible",
        "lithium plating porosity change": "true",  # alias for "SEI porosity change"
    }
)

param = pybamm.ParameterValues("OKane2022")

cycle_number = 1000
exp = pybamm.Experiment(
    ["Hold at 4.2 V until C/100",
    "Rest for 4 hours",
    "Discharge at 0.1C until 2.5 V",  # initial capacity check
    "Charge at 0.3C until 4.2 V",
    "Hold at 4.2 V until C/100",]
    + [("Discharge at 1C until 2.5 V",  # ageing cycles
    "Charge at 0.3C until 4.2 V",
    "Hold at 4.2 V until C/100",)] * cycle_number
    + ["Discharge at 0.1C until 2.5 V"],  # final capacity check
    period="5 minutes"
)

sim = pybamm.Simulation(model, parameter_values=param, experiment=exp)

# safe solver
safe_solver = pybamm.CasadiSolver(atol=1e-3, rtol=1e-3, mode="safe", dt_max=100)

sol = sim.solve(solver=safe_solver)

```

The error message is:

At t = 299.979 and h = 4.56806e-09, the corrector convergence failed repeatedly or with |h| = hmin.
2023-08-07 08:57:26.287 - [ERROR] callbacks.on_experiment_error(224): Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.
ERROR:pybamm.logger:Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.

brosaplanella, 2023-08-07:
Using the following solver works for me:
```python3
safe_solver = pybamm.CasadiSolver(mode="safe")
```

Sunnyturicum, 2023-08-07:
Thank you for your answer. I ran the model with your solver. Unfortunately, I still get the following error message after 319/1000 cycles. 

"""
At t = 299.973 and h = 1.45273e-09, the corrector convergence failed repeatedly or with |h| = hmin.
2023-08-07 19:28:09.253 - [ERROR] callbacks.on_experiment_error(224): Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.
ERROR:pybamm.logger:Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.
"""

Do you have an idea why this happens? From the IDAS documentation, it is unclear to me what causes the error IDA_CONV_FAIL.

brosaplanella, 2023-08-08:
That's strange, mine ran all 1000 cycles. Check that you are using the most recent version of PyBaMM and Casadi.

Sunnyturicum, 2023-08-08:
By separately installing the Casadi Solver, the new version got installed, and it works now - thank you.

```
%pip install casadi
%pip install pybamm
```

flag2freefish, 2023-11-29:
same problem, and try all above methods, and then still failed.
code like this. please help me. thank you very much
import pybamm
import matplotlib.pyplot as plt

model = pybamm.lithium_ion.DFN(
    {
        "SEI": "solvent-diffusion limited",
        "SEI porosity change": "true",
        "lithium plating": "partially reversible",
        "lithium plating porosity change": "true",
        "calculate discharge energy": "true",
    }
)

parameter_values = pybamm.ParameterValues("OKane2022")

def graphite_ocp(sto):
    u_eq = (
        0.6379
        + 0.5416 * pybamm.exp(-305.5309 * sto)
        + 0.044 * pybamm.tanh(-(sto - 0.1958) / 0.1088)
        - 0.1978 * pybamm.tanh((sto - 1.0571) / 0.0854)
        - 0.6875 * pybamm.tanh((sto + 0.0117) / 0.0529)
        - 0.0175 * pybamm.tanh((sto-0.5692) / 0.0875)
    )
    return u_eq

def LFP_ocp(sto):
    u_eq = (
        3.4323
        - 0.8428 * pybamm.exp(-80.2493 * (1 - sto) ** 1.3198)
        - 3.2474e-6 * pybamm.exp(20.2645 * (1 - sto) ** 3.8003)
        + 3.2482e-6 * pybamm.exp(20.2646 * (1 - sto) ** 3.7995)
    )
    return u_eq

def negative_electrode_OCP_entropic_change(sto, c_s_max):
    dEdT = (
        2.8957
        - 5.2972 * pybamm.cos(pybamm.arctan(1/(sto**2) - 1))
        + 4.0370 * pybamm.cos(2 * pybamm.arctan(1/(sto**2) - 1))
        - 2.5304 * pybamm.cos(3 * pybamm.arctan(1/(sto**2) - 1))
        + 1.2717 * pybamm.cos(4 * pybamm.arctan(1/(sto**2) - 1))
        - 0.4884 * pybamm.cos(5 * pybamm.arctan(1/(sto**2) - 1))
        + 0.1301 * pybamm.cos(6 * pybamm.arctan(1/(sto**2) - 1))
        - 0.1086 * pybamm.cos(7 * pybamm.arctan(1/(sto**2) - 1))
    )
    return dEdT

def positive_electrode_OCP_entropic_change(sto, c_s_max):
    dEdT = (
        -0.3538 * sto ** 8
        + 1.3902 * sto ** 7
        - 2.2585 * sto ** 6
        + 1.9635 * sto ** 5
        - 0.9872 * sto ** 4
        + 0.2886 * sto ** 3
        - 0.0463 * sto ** 2
        + 0.0032 * sto
        - 1.9186e-5
    )
    return dEdT

def electrolyte_diffusivity_arrhenius(c_e, T):
    D_c_e = (
        5.2969e-20 * c_e ** 3
        - 2.7984e-16 * c_e ** 2
        + 2.8086e-13 * c_e
        + 3.4857e-10
[... 90 lines cut ...]
parameter_values["Positive electrode porosity"] = 0.336
parameter_values["Separator porosity"] = 0.42
parameter_values["Negative electrode thickness [m]"] = 5.3e-5
parameter_values["Positive electrode thickness [m]"] = 8.5e-5
parameter_values["Separator thickness [m]"] = 1.6e-5
parameter_values["Negative particle radius [m]"] = 8.1e-6
parameter_values["Positive particle radius [m]"] = 1.25e-6
parameter_values["Negative electrode diffusivity [m2.s-1]"] = graphite_diffusivity
parameter_values["Positive electrode diffusivity [m2.s-1]"] = LFP_diffusivity
parameter_values["Negative electrode exchange-current density [A.m-2]"] = graphite_electrolyte_exchange_current_density
parameter_values["Positive electrode exchange-current density [A.m-2]"] = LFP_electrolyte_exchange_current_density
parameter_values["Electrode width [m]"] = 0.47
var_pts = {"x_n": 5, "x_s": 5, "x_p": 5, "r_n": 30, "r_p": 30,}
pybamm.set_logging_level("NOTICE")

cycle_number = 1000
experiment = pybamm.Experiment(
    ["Rest for 1 hours", "Discharge at 0.1C until 2.5 V", "Charge at 1C until 3.65 V", "Hold at 3.65 V until C/100",]
    + [("Discharge at 1C until 2.5 V", "Charge at 1C until 3.65 V", "Hold at 3.65 V until C/25",)] * cycle_number
    + ["Discharge at 0.1C until 2.5 V", "Charge at 1C until 3.65 V", "Hold at 3.65 V until C/100",],
    termination="70% capacity"
)

simulation = pybamm.Simulation(model, parameter_values = parameter_values, experiment = experiment, var_pts = var_pts)
solution = simulation.solve(pybamm.CasadiSolver(mode="safe without grid", return_solution_if_failed_early=True))

flag2freefish, 2023-11-30:
using pybamm>23.5 will output log like they said. it might be a bug.

GeXu66, 2023-12-20:
> By separately installing the Casadi Solver, the new version got installed, and it works now - thank you.
> 
> ```
> %pip install casadi
> %pip install pybamm
> ```

hello! I met the same problem, could you tell me your casadi and pybamm version? Thank you!

Sunnyturicum, 2024-01-11:
> > By separately installing the Casadi Solver, the new version got installed, and it works now - thank you.
> > ```
> > %pip install casadi
> > %pip install pybamm
> > ```
> 
> hello! I met the same problem, could you tell me your casadi and pybamm version? Thank you!

Look at this: https://github.com/ArnaudOlt/dissertation_pybamm