pybamm-team/PyBaMM #1494: Current Function [A] not equals to the Current[A]

gyouhoc, 2021-05-25:
Hi 

I have cases when the input current, 'Current Function [A]' doesn't equal to the output 'Current[A]'. 

For this, I set Current Function [A] = Typical Current [A]. 

[image]

[image]

tinosulzer, 2021-05-25:
Can you post a full example where this happens?

gyouhoc, 2021-05-25:
Please find the script and variables I used for the simulation. 
Although I put the lower voltage limit @ 2.5V, the output voltage is negative. 
Also, the current output and input are different. 

========================================================
parameter_operation['Typical current [A]']= parameter_operation['Current function [A]']
parameter_operation['Initial outer SEI thickness [m]']= parameter_operation['Initial inner SEI thickness [m]']
parameter_operation['Lower voltage cut-off [V]'] = 2.5
parameter_operation['Upper voltage cut-off [V]'] = 100
parameter_operation["Positive electrode active material volume fraction"] = 1-parameter_operation["Positive electrode porosity"] - parameter_operation["DESIGN_CATHODE_VF_IM"]
num = parameter_operation["Positive electrode thickness [m]"]*parameter_operation["Maximum concentration in positive electrode [mol.m-3]"]*(1-parameter_operation["DESIGN_CATHODE_STOICHMIN"])
den = parameter_operation["Negative electrode thickness [m]"]*parameter_operation["Maximum concentration in negative electrode [mol.m-3]"]*(1-0)
parameter_operation["Positive electrode active material volume fraction"] = 1-parameter_operation["Positive electrode porosity"] - parameter_operation["DESIGN_CATHODE_VF_IM"]
parameter_operation["Negative electrode active material volume fraction"] = parameter_operation["Positive electrode active material volume fraction"]*parameter_operation["DESIGN_CELL_NP_RATIO"]*(num/den) 
parameter_operation["Initial concentration in negative electrode [mol.m-3]"] = parameter_operation["DESIGN_ANODE_STOICHMAX"]*parameter_operation["Maximum concentration in negative electrode [mol.m-3]"]
parameter_operation["Initial concentration in positive electrode [mol.m-3]"] = parameter_operation["DESIGN_CATHODE_STOICHMIN"]*parameter_operation["Maximum concentration in positive electrode [mol.m-3]"] 
    #Run the simulation
try:
    model = pybamm.lithium_ion.DFN({"thermal": "x-full","sei":"constant"})
    sim = pybamm.Simulation(model, parameter_values=parameter_operation)
    timescale = parameter_operation.evaluate(model.timescale)
    t_end = timescale
    dt_max_default = 0.001 * timescale
    #sim.specs(parameter_values=parameter_operation)
    sim.solve(t_eval=np.arange(0, 3601, 1),
    solver=pybamm.CasadiSolver(mode="old safe", rtol=1e-9, atol=1e-9,dt_max=dt_max_default/500),)
        #saving sample files
    today = date.today()
    file_name = str(today)+'-'+'-test-number-'+str(x)+'.csv'
            # output file header setup
    sim.solution.save_data(
                    'op_file.csv',
                    [
                        "Time [s]",
                        "Terminal voltage [V]",
                        "Current [A]",
                        "X-averaged cell temperature [K]",
                        "X-averaged negative electrode open circuit potential [V]",
                        "X-averaged positive electrode open circuit potential [V]"
                    ],
                      to_format="csv",
    )   

{'1 + dlnf/dlnc': 2.115343,
 'Activation_energy_electrolyte': 34700,
 'Activation_energy_electrolyte_conductivity': 14251.15,
 'Ambient temperature [K]': 298.15,
 'Bulk solvent concentration [mol.m-3]': 2636.0,
 'Cation transference number': 0.2594,
 'Cell capacity [A.h]': 5.0,
 'Current function [A]': 32.55599999999998,
 'DESIGN_ANODE_DENSITY_HOST': 2266,
 'DESIGN_ANODE_MOLWT_HOST': 0.072064,
 'DESIGN_ANODE_NLI_HOST': 0.727,
 'DESIGN_ANODE_STOICHMAX': 0.8295755,
 'DESIGN_ANODE_VF_IM': 0.0351,
 'DESIGN_CATHODE_DENSITY_HOST': 4800,
 'DESIGN_CATHODE_MOLWT_HOST': 0.088945,
 'DESIGN_CATHODE_NLI_HOST': 0.75,
 'DESIGN_CATHODE_STOICHMAX': 1,
 'DESIGN_CATHODE_STOICHMIN': 0.261772,
 'DESIGN_CATHODE_VF_IM': 0.0441012,
 'DESIGN_CELL_ICL_FRAC_CATUSE': 0.01219512,
 'DESIGN_CELL_NP_RATIO': 1.152,
 'EC diffusivity [m2.s-1]': 2e-18,
 'EC initial concentration in electrolyte [mol.m-3]': 4541.0,
 'Edge heat transfer coefficient [W.m-2.K-1]': 0.3,
 'Electrode height [m]': 0.5611276000000001,
 'Electrode width [m]': 6.80846,
 'Electrolyte conductivity [S.m-1]': <function electrolyte_conductivity_Nyman2008_updated at 0x0000029B66A81438>,
[... 325 lines cut ...]
 'SEI resistivity [Ohm.m]': 4657995.342999999,
 'Separator Bruggeman coefficient (electrode)': 3.7676749999999997,
 'Separator Bruggeman coefficient (electrolyte)': 1.95296,
 'Separator density [kg.m-3]': 757.0,
 'Separator porosity': 0.19522899999999999,
 'Separator specific heat capacity [J.kg-1.K-1]': 700.0,
 'Separator thermal conductivity [W.m-1.K-1]': 0.19,
 'Separator thickness [m]': 9.154045e-05,
 'Typical current [A]': 32.55599999999998,
 'Typical electrolyte concentration [mol.m-3]': 1000.0,
 'Upper voltage cut-off [V]': 100,
 'anode_D_ref': 8.83712716288e-12,
 'anode_E_D_s': 61839.46,
 'anode_E_ex': 17054.83,
 'anode_m_ref': 9.03795,
 'c1': 7.245999999999995,
 'c2': 81.38,
 'c3': -38.141999999999996,
 'c4': -38.98199999999999,
 'c5': -9.846746,
 'cathode_D_ref': 1.8570311429700003e-11,
 'cathode_E_D_s': 86694.40000000001,
 'cathode_E_ex': 78337.81,
 'cathode_m_ref': 2.55762,
 'cathode_stretch': 1}

tinosulzer, 2021-05-26:
Please provide an entire script that can be run. `parameter_operation` is not defined here

gyouhoc, 2021-05-27:
Sorry, I just found a flaw in my script. This leads to save the current function [A] from a different array. This is my mistake.

closed 2021-05-27 by gyouhoc