pybamm-team/PyBaMM #1498: You can do that using post-processed variables after you have solved your `simulation`. Then, you can define your post-processed variable as

ssingh-ipa, 2021-06-02:
You can do that using post-processed variables after you have solved your `simulation`. Then, you can define your post-processed variable as
```python3
c_n_surf = simulation.solution["Negative particle surface concentration [mol.m-3]"]
```
and you can find all the variables you can call in [this notebook](https://colab.research.google.com/github/pybamm-team/PyBaMM/blob/master/examples/notebooks/Getting%20Started/Tutorial%203%20-%20Basic%20plotting.ipynb).

This returns an object `c_n_surf` which is an interpolant. Then, we can call that object and evaluate it at any point by doing
```python3
c_n_surf(x = 1e-5, t = 100)
```

Because this is a surface concentration, it only depends on `x` and `t`. If we chose the concentration in the particles it would depend on `r`, `x` and `t` and thus we would have to evaluate it for the three variables. Note that you can pass time as a vector to evaluate the variable at multiple times.

_Originally posted by @brosaplanella in https://github.com/pybamm-team/PyBaMM/discussions/1295#discussioncomment-200716_

tinosulzer, 2021-06-02:
Did you mean to open this as an issue?

closed 2021-06-03 by tinosulzer

ssingh-ipa, 2021-06-03:
I meant top pose questions on a resolved issue.