pybamm-team/PyBaMM #1500: Print parameter info by submodel

tinosulzer, 2021-06-02:
When printing parameter info, add an option to separate by submodel name (with repetitions between submodels).
This might require storing the parameters when the submodel is created.

TomTranter, 2022-08-10:
Be good to also add a table that lists parameters with a tick against each submodel

rtimms, 2022-09-23:
Note you can do 

```python3
import pybamm

model = pybamm.lithium_ion.SPM()
submodel = model.submodels["positive primary particle"]
submodel.print_parameter_info()
```

jaskiratsingh2000, 2023-02-20:
HI @tinosulzer Can you assign this to me? I'd be more than happy to work on this.

jaskiratsingh2000, 2023-02-20:
Hi @rtimms I see you already implemented the print_parameter_info() above. DO i have to look for the implementation into this page - https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/base_submodel.py
right?

rtimms, 2023-02-20:
At the moment you can do `model.print_parameter_info()` to print the parameters used in the whole model. And you can manually print the parameters for each submodel as in the example above. We want to add the option to do `model.print_parameter_info(by_submodel=True)` which would then print the name of each submodel and the parameters used by that submodel. The method is defined [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/base_model.py#L365).

jaskiratsingh2000, 2023-02-21:
Just bit of clarification. With the help of above mentiioned example. I tried printing. 
By using `submodel.print_parameter_info()`, I get the following:
```
Maximum concentration in negative electrode [mol.m-3] (Parameter)
Reference temperature [K] (Parameter)
Electrode height [m] (Parameter)
Typical current [A] (Parameter)
Separator thickness [m] (Parameter)
Positive electrode thickness [m] (Parameter)
Negative electrode thickness [m] (Parameter)
Number of electrodes connected in parallel to make a cell (Parameter)
Electrode width [m] (Parameter)
Maximum concentration in positive electrode [mol.m-3] (Parameter)
Current function [A] (FunctionParameter with input(s) 'Time [s]')
Positive electrode diffusivity [m2.s-1] (FunctionParameter with input(s) 'Positive particle stoichiometry', 'Temperature [K]')
Positive particle radius [m] (FunctionParameter with input(s) 'Through-cell distance (x) [m]')
Positive electrode active material volume fraction (FunctionParameter with input(s) 'Through-cell distance (x) [m]')
Initial concentration in positive electrode [mol.m-3] (FunctionParameter with input(s) 'Radial distance (r) [m]', 'Through-cell distance (x) [m]')
```

On using `model.print_parameter_info()`, I get the following
```
Maximum concentration in negative electrode [mol.m-3] (Parameter)
Upper voltage cut-off [V] (Parameter)
Typical electrolyte concentration [mol.m-3] (Parameter)
Initial temperature [K] (Parameter)
Negative electrode electrons in reaction (Parameter)
Number of cells connected in series to make a battery (Parameter)
Positive electrode electrons in reaction (Parameter)
Negative electrode Bruggeman coefficient (electrolyte) (Parameter)
Separator Bruggeman coefficient (electrolyte) (Parameter)
Positive electrode Bruggeman coefficient (electrolyte) (Parameter)
Reference temperature [K] (Parameter)
Electrode height [m] (Parameter)
Typical current [A] (Parameter)
Nominal cell capacity [A.h] (Parameter)
Separator thickness [m] (Parameter)
Positive electrode thickness [m] (Parameter)
Negative electrode Bruggeman coefficient (electrode) (Parameter)
Lower voltage cut-off [V] (Parameter)
Negative electrode thickness [m] (Parameter)
Number of electrodes connected in parallel to make a cell (Parameter)
Electrode width [m] (Parameter)
Maximum concentration in positive electrode [mol.m-3] (Parameter)
Positive electrode Bruggeman coefficient (electrode) (Parameter)
Positive electrode porosity (FunctionParameter with input(s) 'Through-cell distance (x) [m]')
Initial concentration in negative electrode [mol.m-3] (FunctionParameter with input(s) 'Radial distance (r) [m]', 'Through-cell distance (x) [m]')
Negative electrode active material volume fraction (FunctionParameter with input(s) 'Through-cell distance (x) [m]')
Positive particle radius [m] (FunctionParameter with input(s) 'Through-cell distance (x) [m]')
Positive electrode OCP [V] (FunctionParameter with input(s) 'Positive particle stoichiometry')
Positive electrode OCP entropic change [V.K-1] (FunctionParameter with input(s) 'Positive particle stoichiometry', 'Maximum positive particle surface concentration [mol.m-3]')
Negative electrode diffusivity [m2.s-1] (FunctionParameter with input(s) 'Negative particle stoichiometry', 'Temperature [K]')
Current function [A] (FunctionParameter with input(s) 'Time [s]')
Negative electrode OCP [V] (FunctionParameter with input(s) 'Negative particle stoichiometry')
Negative electrode porosity (FunctionParameter with input(s) 'Through-cell distance (x) [m]')
Positive electrode active material volume fraction (FunctionParameter with input(s) 'Through-cell distance (x) [m]')
Negative electrode OCP entropic change [V.K-1] (FunctionParameter with input(s) 'Negative particle stoichiometry', 'Maximum negative particle surface concentration [mol.m-3]')
Positive electrode diffusivity [m2.s-1] (FunctionParameter with input(s) 'Positive particle stoichiometry', 'Temperature [K]')
Negative electrode exchange-current density [A.m-2] (FunctionParameter with input(s) 'Electrolyte concentration [mol.m-3]', 'Negative particle surface concentration [mol.m-3]', 'Maximum negative particle surface concentration [mol.m-3]', 'Temperature [K]')
Separator porosity (FunctionParameter with input(s) 'Through-cell distance (x) [m]')
Positive electrode exchange-current density [A.m-2] (FunctionParameter with input(s) 'Electrolyte concentration [mol.m-3]', 'Positive particle surface concentration [mol.m-3]', 'Maximum positive particle surface concentration [mol.m-3]', 'Temperature [K]')
Ambient temperature [K] (FunctionParameter with input(s) 'Time [s]')
Negative particle radius [m] (FunctionParameter with input(s) 'Through-cell distance (x) [m]')
Initial concentration in positive electrode [mol.m-3] (FunctionParameter with input(s) 'Radial distance (r) [m]', 'Through-cell distance (x) [m]')
```

So, I want to knoiw that when we write the `model.print_parameter_info(by_submodel=True)` so it should also print the submodel info as stated above? Correct me if I'm saying wrong.
Looking forward to hearing frop you. @rtimms @tinosulzer @TomTranter

jaskiratsingh2000, 2023-02-22:
Hi @rtimms Any update on this yet?

rtimms, 2023-02-23:
The output of `model.print_parameter_info(by_submodel=True)` should be equivalent to 
```python3
model = pybamm.lithium_ion.SPM()

for name, submodel in model.submodels.items():
    print(name, "submodel parameters:")
    submodel.print_parameter_info()
```

cringeyburger, 2023-12-03:
Hello! I have tried solving this issue by implementing @rtimms method of printing the parameters submodel-wise in my code.

> The output of `model.print_parameter_info(by_submodel=True)` should be equivalent to
> 
> ```python
> for name, submodel in model.submodels.items():
>     print(name, "submodel parameters:")
>     submodel.print_parameter_info()
> ```

```
    def print_parameter_info(self, by_submodel=False):
        """Prints all the extracted parameter information of the model(s)"""
        try:
            if by_submodel:
                if not hasattr(self, 'submodels'):
                    print("ERROR: Model has no submodels.")
                    return

                for submodel_name, submodel in self.submodels.items():
                    print(f"'{submodel_name}' submodel parameters:")
                    submodel.print_parameter_info()
            else:
                self._parameter_info = ""
                parameter_types = [
                    ("Parameter", pybamm.Parameter),
                    ("inputParameter", pybamm.InputParameter),
                    ("FunctionParameter", pybamm.FunctionParameter),
                ]

                for param_type, param_class in parameter_types:
                    parameter_info = self.parameter_info(param_class, param_type)
                    if parameter_info is not None:
                        self._parameter_info += parameter_info
                    else:
                        print(f"WARNING: parameter_info is NONE for {param_type}")

                print(self._parameter_info)
        except Exception as e:
            print(f"ERROR in print_parameter_info: {e}")
```

The feature works as intended, and has passed the pre-commit checks. 
NOTE: To know more about the code, please refer to my [PR](https://github.com/pybamm-team/PyBaMM/pull/3584).