pybamm-team/PyBaMM #1504: add lithium negative electrode

lmolel, 2021-06-05:
**Summary**
 Add model for lithium metal anode.

 **Motivation**
 Lithium metal anodes are attractive because of their high theoretical capacity. Cells with lithium metal anodes are plagued with various chemo-mechanical issues. A PyBaMM model with lithium metal anodes will allow researchers to optimize parameters and ultimately add submodels.

**Additional context**

lmolel, 2021-06-05:
@tinosulzer I'll work on this.

molel-gt, 2021-06-05:
I was logged in differently. Please assign to @molel-gt

tinosulzer, 2021-06-05:
What exactly are you planning to do here? Just trying to minimize duplication of work.

molel-gt, 2021-06-05:
The goal is to be able to run the DFN model for any all-solid-state battery given parameters. I will begin by editing the half-cell model to make it appropriate for lithium metal. Do you think for this case we need a separate full model called `lithium_metal` in addition to existing models `lithium_ion` and `lead_acid`? I am asking because the geometry is slightly different. Lithium ion batteries have electrolytes on both sides of the separator whereas the lithium/polymer/insertion cell has a separator between the composite cathode and the lithium metal anode.

tinosulzer, 2021-06-05:
It probably makes sense to have a separate full model for `lithium_metal`, but it we will need to make sure to avoid code duplication as much as possible - i.e. definitely re-using submodels where possible. The submodels are useful not only for degradation but also for things like comparing different simplifications of the particle model. We should be able to get around the difference in geometry by having dummy variables in the negative electrode

> the default options for the pybamm implementation will not include any submodels. Once the basic version is done, we’ll coordinate on the submodels. What do you think?

Sounds good

molel-gt, 2021-06-05:
Perfect!

molel-gt, 2021-06-14:
I think my current density was too large. I reduced it and now I can get it to work.

molel-gt, 2021-06-14:
As part of this ticket, I would like to get the model to work with the Simulation and Experiment classes. I searched through the existing issues but could not find one that’s doing this. The closest was https://github.com/pybamm-team/PyBaMM/issues/1303 but I am not sure if making the model work with Simulation and Experiment classes are part of the scope of the issue.

molel-gt, 2021-06-16:
Sweet!

molel-gt, 2021-06-16:
One quick thing though. There is still an exception raised when you call the simulation class
`Traceback (most recent call last):
  File "./dfn_half_cell.py", line 102, in <module>
    sim = pybamm.Simulation(model=model, experiment=experiment, parameter_values=params)
  File "/usr/local/lib/python3.8/dist-packages/pybamm/simulation.py", line 101, in __init__
    raise NotImplementedError(
NotImplementedError: BasicDFNHalfCell is not compatible with Simulations yet.`
when I commented out the code and build the package manually, the simulation class worked.
Additionally, it looks like a parameter with default value is no longer required:
`Traceback (most recent call last):
  File "./dfn_half_cell.py", line 103, in <module>
    sim.solve([0, 900])
  File "/usr/local/lib/python3.8/dist-packages/pybamm/simulation.py", line 632, in solve
    self.build(check_model=check_model)
  File "/usr/local/lib/python3.8/dist-packages/pybamm/simulation.py", line 497, in build
    self.set_parameters()
  File "/usr/local/lib/python3.8/dist-packages/pybamm/simulation.py", line 470, in set_parameters
    self._model_with_set_params = self._parameter_values.process_model(
  File "/usr/local/lib/python3.8/dist-packages/pybamm/parameters/parameter_values.py", line 426, in process_model
    model = unprocessed_model.new_empty_copy()
  File "/usr/local/lib/python3.8/dist-packages/pybamm/models/full_battery_models/base_battery_model.py", line 693, in new_empty_copy
    new_model = self.__class__(name=self.name, options=self.options, build=False)
`
when I removed the `build=False` parameter, my simulation worked.

If you'd prefer, I can push the changes. However, I have not tested to make sure other parts of the code don't break.

molel-gt, 2021-06-16:
I have pushed a fix here: https://github.com/pybamm-team/PyBaMM/pull/1514. There are failures unrelated to what I just changed.

tinosulzer, 2021-09-09:
I think this can be closed now that #1600 is merged?

molel-gt, 2021-09-09:
Yes

closed 2021-09-10 by tinosulzer