pybamm-team/PyBaMM #1507: Voltage control when using "surface form"

rtimms, 2021-06-10:
If you to use voltage control with the "surface form" option anything other than "false" you get the error `ModelError: each variable in the algebraic eqn keys must appear in the eqn` due to the check [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/base_model.py#L655). If you comment out the check then it works fine. Should we relax this check further? 

The problem, in this case, is that the "Total current density" variable doesn't appear in the equation `V - pybamm.FunctionParameter("Voltage function [V]", {"Time [s]": pybamm.t * self.param.timescale})`

Full example:
 
```python3
import pybamm 

model = pybamm.lithium_ion.SPM(options={"surface form": "algebraic"})

for var in [
    "Terminal voltage", 
    "X-averaged positive electrode surface potential difference", 
    "X-averaged negative electrode surface potential difference",
]:
    print(model.variables[var])

#experiment = pybamm.Experiment(["Hold at 3.8 V until 10 mA"])
experiment = pybamm.Experiment(
    [
        (
            "Discharge at C/5 for 10 hours or until 3.3 V",
            "Rest for 1 hour",
            "Charge at 1 A until 4.1 V",
            "Hold at 4.1 V until 10 mA",
            "Rest for 1 hour",
        ),
    ]
)

sim = pybamm.Simulation(model, experiment=experiment)
sim.solve()
```

rtimms, 2021-06-10:
I'm not sure why we need this check anyway? So long as the variable appears somewhere in the model then it should be fine? E.g. the model

```python3
model.algebraic = {
    a: a - 2 * c,
    b: b + 3,
    c: a + b -2,
}
````

is valid even though `c` doesn't appear in the equation for `c`. Sure I could swap the keys `a` and `c`, but I shouldn't have to.

tinosulzer, 2021-06-10:
Yeah we don't *need* the check, and there are other checks that would catch most cases that would cause problems anyway. Let's get rid of it

PR #1510 (2021-06-14): #1507 remove algebraic equation check
# Description
Remove the check "each variable in the algebraic eqn keys must appear in the eqn"

Fixes #1507 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue [... cut]
closed 2021-06-14 by rtimms