pybamm-team/PyBaMM #1513: Implementing very long drive cycle to solve DFN model

sful, 2021-06-14:
I'm trying to implement a drive cycle to solve the basic DFN half-cell model. There are around 5000 data points in the drive cycle, but whenever I try to use the current interpolator and then solve the model it gives a runtime error.  I am discretizing the model and then using the Casadi solver in the fast mode. The model can only be solved when I compress the data to around ~1300 points, but then the current interpolator is very inaccurate. Any suggestions on fixing the runtime error so I can pass the entire dataset into the current interpolator and then solve the half-cell model?

tinosulzer, 2021-06-15:
Can you try using casadi solver in safe mode?

sful, 2021-06-15:
Yes, I tried it in "safe" mode a couple times and it keeps crashing. The drive cycle is around 50,000 seconds long, if that helps.

tinosulzer, 2021-06-16:
Can you write a SPM half cell model and see how the drive cycle does? Would be good to make sure it doesn't go outside the voltage limits with the parameters you are using.

sful, 2021-06-18:
Ok, could you help with what parts of the DFN half cell code I have to change to do a SPM half cell model?

tinosulzer, 2021-06-18:
Basically DFN half cell is to basic DFN what SPM half cell would be to basic SPM. Maybe @molel-gt can help with this?

molel-gt, 2021-06-20:
I still have not been able to run long cycles in Basic DFN half cell. Without measuring the time, I noticed that I could not run a simulation for 10 hours using the DFN half cell, but could do that with the full DFN model. The other thing would be to check the current density. The model fails during discharge for large current densities due to polarization.

sful, 2021-06-21:
Ok, I will check current density. Have you ever tried running a long cycle in a simpler half-cell model (SPM or SPMe)?

tinosulzer, 2021-11-13:
@sful can you try this with the recently updated models in https://github.com/pybamm-team/PyBaMM/blob/develop/examples/scripts/compare_lithium_ion_half_cell.py (without the SEI option)

tinosulzer, 2022-12-13:
Not enough details to fix, closing

closed 2022-12-13 by tinosulzer