pybamm-team/PyBaMM #1519: failure when solving using simulation class + specified experiment

karthikmayil, 2021-06-22:
Trying to reproduce [this example notebook](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/Creating%20Models/5-half-cell-model.ipynb), and was able to with no trouble (aside from one small fix). However, when trying to solve the exact same model under cycling protocol using the simulation class, I run into problems.

**To Reproduce**
1. Start with [example notebook](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/Creating%20Models/5-half-cell-model.ipynb), replacing `concatenation` with `Concatenation`
2. Add nominal capacity `'Nominal cell capacity [A.h]':0.9` to parameter values `param`
3. Solve the model using a specified experimental protocol and simulation class

```
protocol = pybamm.Experiment(
    [
        ("Charge at 1C until 4.3 V",
        "Hold at 4.3 V until C/5",
        "Discharge at 1C until 3 V",
        "Rest for 1 hour")
    ]
)
sim = pybamm.Simulation(model, experiment=protocol, parameter_values=param)
solution = sim.solve()
```
which returns the following:
`AttributeError: 'Subtraction' object has no attribute '_saved_evaluate_for_shape'`
and
`ValueError: operands could not be broadcast together with shapes (140,1) (20,1)`
when setting up the experiment.

The  `(20,1)` comes from the 20 points used to discretize the positive electrode, as far as I can tell. I can't tell where the 140 is from. 

Let me know what you think, thanks!

rtimms, 2021-06-22:
At the moment this model isn't set up to work with experiments and is just there to show you how to go about setting up models which have multiple domains. I've had a bit of a dig and it looks like it's related to how we set up models for running experiments (we make multiple copies for current, voltage, power control and switch between them for the different steps). The issue is because this notebook just uses a `pybamm.BaseModel` rather than `BaseBatteryModel`.

[This](https://github.com/pybamm-team/PyBaMM/pull/1514) pull request will make it so that you can use the `BasicDFNHalfCell` with the `Simulation` class, and there is an issue open [here](https://github.com/pybamm-team/PyBaMM/issues/1518) that will make the half cell models integrate with pybamm submodels. After that is done you should be able to use more functionality with half cell models. 

If you're interested in using pybamm for half cell models I'd recommend you take a look at the `BasicDFNHalfCell` [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/lithium_ion/basic_dfn_half_cell.py) and keep you're eyes on issue #1518 . If you're feeling confident you might be able to make the necessary changes so that  `BasicDFNHalfCell`  works with experiments! Let us know how you get on.

tinosulzer, 2021-06-22:
By the way, make sure you use the notebook version from the `main` branch if you are testing using the latest pip release, and from the `develop` branch if you are testing using the latest version of develop

tinosulzer, 2021-09-14:
Is this still an issue with the latest version @karthikmayil ?

tinosulzer, 2021-11-09:
Closing but feel free to reopen if still a problem

closed 2021-11-09 by tinosulzer