pybamm-team/PyBaMM #1527: Expose additional parameters in DFN half cell

molel-gt, 2021-06-28:
**Summary**
Expose additional variables/parameters that are already computed, and make them accessible in the solution.

 **Motivation**
Ratio of diffusion times to discharge times are already computed, but not exposed in variables. So is current density.

**Additional context**
Parameters to add include:

- Current density [A.m-2]
- Ratio of electrolyte transport and discharge timescales
- Ratio of solid diffusion and discharge timescales
- Divergence in current density

It would likely be useful to expose all variables that are not user-inputs.

molel-gt, 2021-07-12:
@tinosulzer I would like to pick this up, but wanted to confirm with you first.

tinosulzer, 2021-07-12:
Yep go for it

tinosulzer, 2021-09-10:
Closed since full DFN can now do half-cell simulations

closed 2021-09-10 by tinosulzer