pybamm-team/PyBaMM #1541: Update parameter management notebook

rtimms, 2021-07-09:
Update the parameter management notebook to show how to create a `chemistry` dict from your own parameters, e.g.
like this
```python3
Chen2020 = {
    "chemistry": "lithium_ion",
    "cell": "LGM50_Chen2020",
    "negative electrode": "graphite_Chen2020",
    "separator": "separator_Chen2020",
    "positive electrode": "nmc_Chen2020",
    "electrolyte": "lipf6_Nyman2008",
    "experiment": "1C_discharge_from_full_Chen2020",
    "sei": "example",
    "citation": "Chen2020",
}
```
but passing your own folder names. 

This will make it easier for people who want to add an entirely new set locally, but using the cell, negative electrode, .... file structure.

leomg08, 2021-07-13:
I just wanted to point out that in my Spyder this had worked using ' ' instead of " " for some reason. Using " " returned an error...

leomg08, 2021-07-13:
I also wanted to ask a follow up question on this:
-> Within the custom parameter set folders, it is recommended to have a readme. file. I am not familiar with these. Along with the structure of Chen2020, I have used the readme. files in my custom parameter folders but did not edit these, is this a problem or is this fine?

rtimms, 2021-07-29:
The readme just gives some information on where the parameters came from (e.g. gives references and any other notes)

rtimms, 2022-11-01:
Closing this as parameters are now provided more easily in a single file (see #2342). Third party sets can be added using [entry points ](https://github.com/pybamm-team/PyBaMM/pull/2396).

closed 2022-11-01 by rtimms