pybamm-team/PyBaMM #1547: Solving 2D pouch model with explicit experiment fails

DavidMStraub, 2021-07-14:
**Describe the bug**

Running the default pouch model with a manually specified `Experiment` causes the Casadi solver to fail.

**To Reproduce**

```python
import pybamm
options = {"cell geometry": "pouch", "thermal": "x-lumped", "dimensionality": 2}
model = pybamm.lithium_ion.SPMe(options=options)
experiment = pybamm.Experiment([
    "Discharge with 5 C for 1 minute"
])
sim = pybamm.Simulation(model, experiment=experiment)
sim.solve()
```

**Expected behaviour**
Solve

**Screenshots**

```
RuntimeError: .../casadi/core/function_internal.hpp:1257: Input 1 (i1) has mismatching shape. Got 11-by-1. Allowed dimensions, in general, are:
 - The input dimension N-by-M (here 2-by-1)
 - A scalar, i.e. 1-by-1
 - M-by-N if N=1 or M=1 (i.e. a transposed vector)
 - N-by-M1 if K*M1=M for some K (argument repeated horizontally)
 - N-by-P*M, indicating evaluation with multiple arguments (P must be a multiple of 1 for consistency with previous inputs)
 - ```

tinosulzer, 2021-07-14:
This will fix it for now
```python3
sim.solve(calc_esoh=False)
```
also `solver=pybamm.CasadiSolver("fast with events")` might be a bit faster

closed 2021-07-15 by tinosulzer

DavidMStraub, 2021-09-30:
Hi @tinosulzer, I just ran into a very similar issue (`Input 1 (i1) has mismatching shape. Got 141-by-1.`) when doing a simulation with `"lithium plating": "irreversible"`. Setting `calc_esoh=False` didn't solve it. I'm using current `develop`. Any idea how to fix it?

DavidMStraub, 2021-09-30:
I'll open a new issue with a minimal example.

PR #1548 (2021-07-15): #1547 fix esoh model in pouch cell
# Description

Fixes #1547 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ] All tests pass: `$ pytho [... cut]