pybamm-team/PyBaMM #1553: inquiry on the battery model parameters

gyouhoc, 2021-07-17:
Hi, 

I just want to ask you about the following four parameters. 
Would you explain how these four parameters applied to the P2D model? 

Negative electrode double-layer capacity [F.m-2]
Positive electrode double-layer capacity [F.m-2]

Negative tab width [m] <- is this related to the foil size?
Positive tab width [m]<- is this related to the foil size?

rtimms, 2021-07-19:
The double-layer capacity parameters are if you include capacitance effects in the model (using the option `{"surface form": "differential"}`). In this case, you solve an equation for the potential difference of the form 

[image]

where `C_dl` is the double-layer capacitance. 

The tab width is only relevant for the pouch cell models (with the options `options = {"cell geometry": "pouch", "dimensionality": 2, "current collector": "potential pair"}` and refers to the width of the tabs on the pouch cell.

gyouhoc, 2021-07-19:
I did not use any of these model options but used DFN and SEI options. 
However the cell voltage influenced by the capacitance and battery tab width parameters. 

If I want to make sure that I exclude these effects, what I need to do?

rtimms, 2021-07-20:
Could you post an example to reproduce this and I can take a look?

You can do 
```python3
import pybamm

model = pybamm.lithium_ion.DFN()
model.print_parameter_info()
```
to get a list of all the parameters that show up in the model.

tinosulzer, 2021-09-13:
@gyouhoc any update? Otherwise I will close this

gyouhoc, 2021-09-14:
Sorry for the late response.
Please close the issue.

Thank you

Cho

closed 2021-09-14 by tinosulzer