pybamm-team/PyBaMM #1554: Dimensions interachanged in 2D processed variable

DavidMStraub, 2021-07-19:
**Describe the bug**

I'm plotting a 2D processed variable and finding that when I specify the spatial variables y and z, they seem to be interchanged.

**To Reproduce**
Steps to reproduce the behaviour:

```python
import pybamm
options = {"cell geometry": "pouch", "thermal": "x-lumped", "dimensionality": 2, "current collector": "potential pair"}
model = pybamm.lithium_ion.SPMe(options=options)
experiment = pybamm.Experiment(["Discharge with 3 C for 10 seconds"])
solver = pybamm.CasadiSolver("fast")
sim = pybamm.Simulation(model, experiment=experiment, solver=solver)
sim.solve()
sim.plot(["X-averaged Ohmic heating [W.m-3]"], spatial_unit="m")
print(sim.solution["X-averaged Ohmic heating [W.m-3]"](t=10, y=0.147, z=0.137))
print(sim.solution["X-averaged Ohmic heating [W.m-3]"](t=10, y=0.207, z=0.1))
```

[image]

```
array(9830.35092739)
array(17052.65235706)
```

**Expected behaviour**

The first printed value should be directly on top of the positive tab, which - as expected and shown in the plot - has a larger heat generation. The second printed value, which has the large heat generation, is a (y, z) point on the right edge of the pouch. So it seems that y and z are interchanged.

rtimms, 2021-07-20:
Looks like this was an issue with how the entries of variables that were discretised using FEM were indexed. I have opened a PR with a fix. @DavidMStraub could you checkout the branch and double-check this works as expected please?

DavidMStraub, 2021-07-20:
Yes, looks good (for the case I was considering), thanks!

PR #1556 (2021-07-21): #1554 fix indexing for FEM entries
# Description

Fixes the order of the indexing for the entries of variables solved using FEM

Fixes #1554

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Ke [... cut]
closed 2021-07-21 by rtimms