pybamm-team/PyBaMM #1561: initial_soc ignored with experiment

TomTranter, 2021-07-22:
**Describe the bug**
When running a simulation with an experiment the initial_soc argument seems to be ignored. This may be due to solver.step rather than solver.solve 

**To Reproduce**
```
import pybamm
from pybamm.parameters.parameter_sets import Chen2020

model = pybamm.lithium_ion.SPMe()
param = pybamm.ParameterValues(chemistry=Chen2020)

socs = [0.25, 0.5, 0.75]
# Works as expected
sims = []
for soc_init in socs:
    sim = pybamm.Simulation(model=model, parameter_values=param)
    sim.solve(t_eval=[0, 1000], initial_soc=soc_init)
    sims.append(sim)
pybamm.dynamic_plot(sims, labels=socs)
# Does not account for initial SOC with experiment
esims = []
exp = pybamm.Experiment(['Discharge at 1C until 2.6V (1 minute period)'])
for soc_init in socs:
    sim = pybamm.Simulation(model=model, parameter_values=param, experiment=exp)
    sim.solve(initial_soc=soc_init)
    esims.append(sim)
pybamm.dynamic_plot(esims, labels=socs)
```

**Expected behaviour**
inital_soc is set at start of experiment

TomTranter, 2021-07-22:
@tinosulzer you may already be on top of this but if you like I can take a look. This stands somewhere between bug and feature I guess - should at least throw a warning that it's ignored with experiment for now

TomTranter, 2021-07-26:
I think the issue is that with experiment the parameters are copied to the simulation object in self.op_conds_to_model_and_param in the init with set_up_experiment and then solve only overrides the initial concentration on the self.parameter_values dict so maybe intial_soc needs to be passed to simulation init not the solve method

TomTranter, 2021-07-26:
@tinosulzer I am happy to make the changes if you agree we should move initial_soc to the simulation init

tinosulzer, 2021-07-27:
Ideally it would be in the solve call so that you can easily solve a simulation comparing different initial socs. Otherwise we can put it in the init. Or even in the experiment might be most logical

TomTranter, 2021-07-27:
Yeah I was thinking in experiment but would that be a bit confusing if it were a command that looked like it jumped to an SOC rather than charging or discharging to it. I can see the benefit of having it as part of solve but it would be simpler I think to have in the simulation init. Comparing simulations is just as easy as solutions but takes longer to set up I guess.

PR #1569 (2021-07-27): Get initial_soc working as simulation solve argument with experiments
# Description

Update the stored parameter_values dicts on the simulation when running experiments to set the initial soc

Fixes #1561

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code [... cut]
closed 2021-07-27 by tinosulzer