pybamm-team/PyBaMM #1574: Add citation.cff

tinosulzer, 2021-07-28:
https://twitter.com/natfriedman/status/1420122675813441540

priyanshuone6, 2021-07-28:
This might help https://citation-file-format.github.io/cff-initializer-javascript/

tinosulzer, 2021-07-28:
Nice, can you do this @priyanshuone6 ?

priyanshuone6, 2021-07-28:
Okay

brosaplanella, 2021-07-28:
Also, this might be a good time to create a Zenodo DOI for PyBaMM and link it so a new DOI is provided for each release.

tinosulzer, 2021-07-28:
Kind of prefer people citing the paper than a Zenodo DOI tbh

brosaplanella, 2021-07-28:
I think it is better too, but will the cff to cite the paper or the code? If the latter, it is better to give it a DOI.

priyanshuone6, 2021-07-28:
let me know if there are any changes
```
---
authors: 
  -
    family-names: Sulzer
    given-names: Valentin
  -
    family-names: Marquis
    given-names: Scott G.
  -
    family-names: Timms
    given-names: Robert
  -
    family-names: Robinson
    given-names: Martin
  -
    family-names: Chapman
    given-names: S. Jon
cff-version: "1.1.0"
date-released: 2021-06-08
doi: "10.5334/jors.309"
keywords: 
  - battery modelling
  - expression tree
  - python
  - symbolic differentiation
license: "BSD 3-Clause"
message: "If you use PyBaMM, please cite it using the following metadata."
repository-code: "https://github.com/pybamm-team/PyBaMM"
title: "Python Battery Mathematical Modelling (PyBaMM)"
...
```

PR #1579 (2021-07-29): Create CITATION.cff
# Description

Create CITATION.cff

Fixes #1574 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ] All tests pass: `$ [... cut]
closed 2021-07-29 by tinosulzer