pybamm-team/PyBaMM #1575: Add Discharged energy [W.h] as variable

TomTranter, 2021-07-28:
**Summary**
Add Discharged energy [W.h] as variable

 **Motivation**
Like discharged capacity but using current * voltage. Useful for performance metrics and minimal overhead

tinosulzer, 2021-09-13:
Took a look at how you started this. You don't need to create a new `pybamm.Variable` for discharged energy, just calculate it as a post-processing step

TomTranter, 2021-10-03:
Ok - is there an official way to do this? Or you just mean manually with the entries

tinosulzer, 2021-10-03:
add it to this function somewhere https://github.com/pybamm-team/PyBaMM/blob/51ee22aece7389434750a7f49ed1252fe4a3a9de/pybamm/models/full_battery_models/base_battery_model.py#L828

tinosulzer, 2022-01-19:
My bad, you do need an additional variable

PR #1969 (2022-03-10): Issue 1575 discharge energy
# Description

Add "Discharge energy [W.h]" as an output variable (integral of power).
Set option "calculate discharge energy" to "true" to turn on. It's false by default as calculating it can be slow.

Fixes #1575 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization ( [... cut]
closed 2022-03-10 by tinosulzer