pybamm-team/PyBaMM #1580: Allow variables to have more than 3 domains

tobykirk, 2021-07-29:
**Summary**
 Currently variables can have a maximum of 3 domains, i.e. a `"primary"` domain and two auxiliary domains, `"secondary"` and `"tertiary"`. This could be extended to allow `"quaternary"` etc. domains (or changed to allow any number of domains).

Also, a `TertiaryBroadcast()` (or a generalization) could be defined, similar to `PrimaryBroadcast()` and `SecondaryBroadcast()`.

 **Motivation**
 Necessary in order to add particle-size distributions to the DFN framework, where some variables live on 4 different domains , e.g.
```python
 c_s_distribution = pybamm.Variable(
    "Negative particle concentration distribution",
    domain="negative particle",
    auxiliary_domains={
        "secondary": "negative particle size",
        "tertiary": "negative electrode",
        "quaternary": "current collector",
    },
    bounds=(0, 1),
)
```
(Even if some domains are 0D in practice, they can push others off the domain list before the domains are discretized.)

A `TertiaryBroadcast()` would then be used for e.g. broadcasting the "X-averaged" concentration distribution to the "electrode" domains (which would be added in the `"tertiary"` position).

**Additional context**
 Add any other context or screenshots about the feature request here.

tinosulzer, 2021-08-02:
Not sure `TertiaryBroadcast` is needed. `PrimaryBroadcast` broadcasts from a lower dimension to a higher dimension (e.g. particle to electrode) and `SecondaryBroadcast` broadcasts the other way. The names are definitely misleading, I think originally they were intended differently

tobykirk, 2021-08-02:
I see @tinosulzer. I could be mistaken, but it looks to me that `SecondaryBroadcast` always puts the higher dimension broadcast domain in the secondary position (shifting previous secondary domain to tertiary). For a max of 3 domains, this basically covers all the higher dimension broadcast cases (as variables either already have the current collector as a domain, or no domains at all in which case you'd use `FullBroadcast` instead).

But, a new use-case with the size domain would be, e.g. starting from a symbol with domains

```python
domains = {
    "primary": "negative particle",
    "secondary": "negative particle size",
    "tertiary": "current collector"
}
```
and broadcasting to `"negative electrode"`, ending up with
```python
domains = {
    "primary": "negative particle",
    "secondary": "negative particle size",
    "tertiary": "negative electrode",
    "quaternary": "current collector"
}
```

Is this possible with `SecondaryBroadcast` currently? If not, could change it to put the broadcast domain in an auxiliary position of choice, or just put it where we know it should go (electrode always after particle, etc)? Maybe motivates renaming to something like `AuxiliaryBroadcast`, though.

tobykirk, 2021-08-02:
I'm willing to take a shot at this (adding quaternary domains), but any other suggestions of how to implement this are definitely welcome!

tinosulzer, 2021-08-02:
Oh yeah I think you're right. Secondary broadcast is actually for squeezing in a domain in the 2nd position and moving 2nd position to 3rd position, so we do need a new tertiary. Definitely good to have someone else look at this part of the code!

PR #1590 (2021-08-09): 1580 issue more domains
# Description
Fixes #1580

Extended the domain list of symbols to allow a fourth ("quaternary") domain, and updating the spatial operators that add or remove domains (thus shifting their position up and down) to reflect this: 
 -  `PrimaryBroadcast` and `SecondaryBroadcast`
 -  `Integral`
 -  `BoundaryValue`

Also added a `TertiaryBroadcast` class, similar to `SecondaryBroadcast` but puts the broadcast domain in the `"tertiary"` rather than `"secondary"` position.

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-t [... cut]
closed 2021-08-09 by tinosulzer