pybamm-team/PyBaMM #1599: UDDS loading profile

harish2297, 2021-08-09:
Is there support for UDDS loading profile in pybamm. I can see various researchers using this loading profile in batteries in MATLAB.

tinosulzer, 2021-08-09:
Look for `US06` in this example https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/change-input-current.ipynb. You can replace our US06 cycle with your UDDS cycle. If you could add the UDDS cycle [here](https://github.com/pybamm-team/PyBaMM/tree/develop/pybamm/input/drive_cycles) that would be really helpful!

harish2297, 2021-08-09:
@tinosulzer Thank you for your response. I will check out the example. Sorry I don't have the UDDS cycle profile with me. Hope somebody else contributes to that.

tinosulzer, 2021-08-09:
You will need to provide the UDDS cycle to be able to simulate it

harish2297, 2021-08-09:
Yeah I got it :+1: