pybamm-team/PyBaMM #1605: Add docs and example for plot_voltage_components

rtimms, 2021-08-12:
Add documentation for `pybamm.plot_voltage_components` and either add a new example showing this or add this plot to an existing example

Saransh-cpp, 2021-09-07:
Can I take this up? if no one is working on it.

rtimms, 2021-09-07:
Yep, thanks!

closed 2021-09-09 by tinosulzer

nickbrady, 2024-04-17:
I know this issue is closed, but if I follow the code outlined in [Tutorial 3: Basic Plotting](https://docs.pybamm.org/en/latest/source/examples/notebooks/getting_started/tutorial-3-basic-plotting.html) `sim.plot_voltage_components()` produces the error: AttributeError: 'Simulation' object has no attribute 'plot_voltage_components'

MWE
```python
import pybamm

model = pybamm.lithium_ion.DFN()
sim = pybamm.Simulation(model)
sim.solve([0, 3600])

sim.plot_voltage_components()
```

agriyakhetarpal, 2024-04-17:
Hi, @nickbrady – I tried to reproduce the MWE with the latest changes from the `develop` branch and also on the latest release (v23.9), but I cannot reproduce the issue. The code executes without errors for me and produces the desired plot with all of the overpotentials. Could you please re-install PyBaMM or try it within a fresh environment? If it doesn't work, I would suggest creating a new issue for this.

nickbrady, 2024-04-22:
I reinstalled pybamm today 

pybamm.__version__ = '24.1'

and I get the same error. I am getting errors when going through other snippets of code from the Examples section of the documentation as well.

PR #1645 (2021-09-09): Add documentation and example for `plot_voltage_component`
# Description

 - Added documentation for `plot_voltage_components`
 - Extended the `Basic plotting` example notebook to include `plot_voltage_components`

Fixes #1605 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change  [... cut]