pybamm-team/PyBaMM #1607: SEI option is not recognized in the example script calendar_ageing.py

wigging, 2021-08-15:
**Describe the bug**
I'm trying to run the `calendar_ageing.py` example in `PyBaMM/examples/scripts/` but I get an error that the `SEI` option is not recognized.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to `PyBaMM/examples/scripts/`
2. Run `calendar_ageing.py` from the terminal
3. See error shown below

```
File "/Users/gavinw/miniconda3/envs/bamm/lib/python3.9/site-packages/pybamm/models/full_battery_models/lithium_ion/spm.py", line 33, in __init__
    super().__init__(options, name)
  File "/Users/gavinw/miniconda3/envs/bamm/lib/python3.9/site-packages/pybamm/models/full_battery_models/lithium_ion/base_lithium_ion_model.py", line 17, in __init__
    super().__init__(options, name)
  File "/Users/gavinw/miniconda3/envs/bamm/lib/python3.9/site-packages/pybamm/models/full_battery_models/base_battery_model.py", line 99, in __init__
    self.options = options
  File "/Users/gavinw/miniconda3/envs/bamm/lib/python3.9/site-packages/pybamm/models/full_battery_models/base_battery_model.py", line 222, in options
    raise pybamm.OptionError(
pybamm.expression_tree.exceptions.OptionError: Option 'SEI' not recognised. Best matches are ['sei', 'side reactions', 'convection']
```

**Expected behaviour**
The expected option is `SEI` so I don't know why lower case letters `sei` are given as the best match. Other examples as well as the documentation use `SEI` in capital letters.

rtimms, 2021-08-16:
This is probably related to #1608 - in older versions the option was "sei" and we changed it to "SEI". Can you try installing via pip or from source instead of using conda?

wigging, 2021-08-16:
Yes, due to the conda issue #1608, I was using an older version of PyBaMM. I installed the latest version using pip then ran the calendar_ageing example and now the error is gone. However, another error is now occurring but I will submit a different issue for the new error.

closed 2021-08-16 by wigging