pybamm-team/PyBaMM #1608: Conda does not install latest version of PyBaMM

wigging, 2021-08-16:
**Describe the bug**
I followed the instructions to install PyBaMM using conda but the latest version that it installs is 0.2.3. At the time of writing this issue, the latest version is 0.4.0.

**To Reproduce**
Steps to reproduce the behaviour:
1. Install with `conda install -c conda-forge pybamm`
2. Check version in iPython using `pybamm.__version__`

**Expected behaviour**
Conda should install the latest version of PyBaMM.

tinosulzer, 2021-08-30:
I think @Saransh-cpp  is looking into the conda install?

Saransh-cpp, 2021-09-07:
I think the package's version is not updating in conda-forge. [This](https://anaconda.org/conda-forge/pybamm) shows that the latest version available through conda installation is `0.2.3.post1`. Are the new versions manually uploaded or is there an automatic process of updating the package?

priyanshuone6, 2021-09-07:
The conda-forge is running a web service that will (try to) automatically bump the version of the recipe when a new version is released on PyPI. For some reason, the package is not getting updated in the [pybamm-feedstock](https://github.com/conda-forge/pybamm-feedstock). But we can also update the version manually by making a pull request to the feedstock repository. @tlestang maintains that repo, I think he might know what went wrong?

raghuramshankar, 2021-09-07:
Hi,

I encountered a similar issue when installing using pip. Here is my system config:

Python version:
```
python --version
Python 3.9.6
```

Linux version:
```
uname -r
5.13.12-1-MANJARO
```

When I try to reinstall  `pybamm` after uninstalling, it says the following:
```
pip install pybamm                                                                            ✔ 
Defaulting to user installation because normal site-packages is not writeable
Collecting pybamm
  Using cached pybamm-0.4.0.post1-cp39-cp39-manylinux2010_x86_64.whl (1.5 MB)
Requirement already satisfied: jupyter in ./.local/lib/python3.9/site-packages (from pybamm) (1.0.0)
Requirement already satisfied: matplotlib>=2.0 in ./.local/lib/python3.9/site-packages (from pybamm) (3.4.3)
Requirement already satisfied: scikit-fem>=0.2.0 in ./.local/lib/python3.9/site-packages (from pybamm) (3.2.0)
Requirement already satisfied: pandas>=0.24 in ./.local/lib/python3.9/site-packages (from pybamm) (1.3.2)
Requirement already satisfied: anytree>=2.4.3 in ./.local/lib/python3.9/site-packages (from pybamm) (2.8.0)
  Using cached pybamm-0.4.0-cp39-cp39-manylinux2010_x86_64.whl (1.5 MB)
  Using cached pybamm-0.2.4.post3-py3-none-any.whl (520 kB)
.
.
(more libraries checked)
.
.
Installing collected packages: pybamm
Successfully installed pybamm-0.2.4.post3
```

I think this could mean that there is a cached version of `pybamm 0.4.0.post1` downloaded and available locally on my system, but for some reason it refuses to install it. 
Also, I see that the latest release version of `pybamm` is `v21.08`. This version does not get downloaded nor gets installed on my system.

Saransh-cpp, 2021-09-07:
The older version of pybamm is probably still available in your system somewhere and pip is picking that up. This might help -
https://stackoverflow.com/a/17153977/14746647
https://stackoverflow.com/a/44008863/14746647

`v21.08` is not available on PyPi right now, but you can install it through GitHub - 
```
pip install git+https://github.com/[repo owner]/[repo]@[branch name]
```

tinosulzer, 2021-09-20:
@all-contributors add @wigging for bug, code

wigging, 2021-09-30:
Instead of using conda-forge can PyBaMM be released on its own conda channel? For example, the Cantera package can be installed directly from Anaconda.org using `conda install -c cantera cantera`. https://anaconda.org/Cantera/cantera

tinosulzer, 2021-09-30:
Possibly, yes. @priyanshuone6 is working on this issue. Out of interest, what is the downside of pip installing into a conda environment? I am new to conda

priyanshuone6, 2021-09-30:
Having issues upgrading PyBaMM's conda-forge to v21.8 due to incompatibility issues when using python 3.6, I hope things work out in the next release

tinosulzer, 2021-12-01:
Now fixed, thanks @priyanshuone6 !

closed 2021-12-01 by tinosulzer