pybamm-team/PyBaMM #1610: Key error for calendar_ageing example

wigging, 2021-08-16:
**Describe the bug**
When running the `calendar_ageing.py` example script, a key error occurs where `Loss of lithium inventory [%]` is not found. Another key error occurs for `Total lithium lost [mol]`.

Changing the original key values in the plot function from

```python
pb.dynamic_plot(
    sims,
    [
        "Terminal voltage [V]",
        "Negative particle surface concentration",
        "X-averaged negative particle surface concentration",
        "Electrolyte concentration [mol.m-3]",
        "Total negative electrode SEI thickness [m]",
        "X-averaged total negative electrode SEI thickness [m]",
        "X-averaged total negative electrode SEI thickness",
        "X-averaged negative electrode SEI concentration [mol.m-3]",
        "Loss of lithium to negative electrode SEI [mol]",
        "Sum of x-averaged negative electrode interfacial current densities",
        "Loss of lithium inventory [%]",
        ["Total lithium lost [mol]", "Loss of lithium to negative electrode SEI [mol]"],
    ],
)
```

to the following seems to fix the problem

```python
pb.dynamic_plot(
    sims,
    [
        "Terminal voltage [V]",
        "Negative particle surface concentration",
        "X-averaged negative particle surface concentration",
        "Electrolyte concentration [mol.m-3]",
        "Total negative electrode SEI thickness [m]",
        "X-averaged total negative electrode SEI thickness [m]",
        "X-averaged total negative electrode SEI thickness",
        "X-averaged negative electrode SEI concentration [mol.m-3]",
        "Loss of lithium to negative electrode SEI [mol]",
        "Sum of x-averaged negative electrode interfacial current densities",
    ],
)
```

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to `PyBaMM/examples/scripts/calendar_ageing.py`
2. Run `$ python calendar_ageing.py`
3. See error shown below.

```
  File "/Users/gavinw/miniconda3/envs/bamm/lib/python3.9/site-packages/pybamm/util.py", line 93, in __getitem__
    raise KeyError(f"'{key}' not found. Best matches are {best_matches}")
KeyError: "'Loss of lithium inventory [%]' not found. Best matches are ['Loss of lithium to negative electrode SEI [mol]', 'Loss of lithium to positive electrode SEI [mol]', 'Loss of lithium to negative electrode lithium plating [mol]']"
```

rtimms, 2021-08-16:
I think this is probably a version issue again. The version you get via pip is in line with the `main` branch. The default `develop` branch is ahead of `main`. You can take a look at the `calendar_aging.py` example on the `main` branch [here](https://github.com/pybamm-team/PyBaMM/blob/main/examples/scripts/calendar_ageing.py)

wigging, 2021-08-16:
The example from the `main` branch works fine. Why is the `develop` branch the default branch on GitHub?

wigging, 2021-08-16:
Usually `main` or `master` branch is the default branch which is aligned with the released version of the software or code. Having `develop` as the default branch is confusing especially for users like me who are trying to run the examples with the pip installed version of PyBaMM.

tinosulzer, 2021-08-16:
The reason we have `develop` as the default branch is that this is where PRs should automatically merge into. It's not that much work telling people they need to look at the main branch for the examples if using pip .... we are switching to monthly releases anyway so any non-backwards-compatible changes (of which there are hopefully not too many others) will be short-lived

wigging, 2021-08-17:
@tinosulzer Is it mentioned in the README or documentation about running examples from the main branch?

tinosulzer, 2021-08-17:
Just opened a PR, can you check it @wigging ?

wigging, 2021-08-17:
@tinosulzer Yes, the pull request looks good. Thank you for adding a statement about the examples to the README.

PR #1613 (2021-08-17): Update README.md
Fixes #1610 (clarification about where examples work)

# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes # (issue)

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds  [... cut]
commit 106b8860 (2021-08-17): Update README.md

Fixes #1610 (clarification about where examples work)
closed 2021-08-17 by tinosulzer